esp-rs/esp-hal #2623: Resolve `C-EXAMPLE` violations

bugadani, 2024-11-28:
As documentation isn't as immutable as the API is, we may not treat rustdoc examples as strictly as the Rust API guidelines wants us to do. This issue is meant as a reminder to resolve the guidelines in however we will see fit, before 1.0.

https://rust-lang.github.io/api-guidelines/documentation.html#c-example

Dominaezzz, 2025-06-27:
Perhaps links to centralised docs explaining the concept would be better than repeated docs.

Like the pin setters could say something along the lines of "See the gpio module docs for how these functions work".

MabezDev, 2025-07-04:
> Should we also document `AnyX` `clone_unchecked` and `reborrow` methods? We could do something like we do for `into_async/blocking` have an specific example for any of the structs?
> 
> Same question applies to the config structs we build with the builder lite derive

`AnyX` might be a bit hard to explain globally, but I hope just a written comment on those structs will be enough. For the others, particularly `reborrow` we should link to the top level docs here: https://docs.espressif.com/projects/rust/esp-hal/1.0.0-beta.1/esp32c6/esp_hal/index.html#peripheral-singletons

SergioGasquez, 2025-07-08:
With #3770, I think all our public macros, functions, methods, and macros have examples (or link to examples), the only exceptions are the `Config` structs generated with the builder lite pattern.

PR #3770 (2025-07-09): Add missing c-examples
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [x] I have added n [... cut]