esp-rs/esp-hal #2641: UART driver does not support autobaud functionality

jessebraham, 2024-12-02:


jessebraham, 2024-12-19:
I believe this is just a matter of setting `UART_AUTOBAUD_EN`, so would only require a field in `Config` to enable this or something. However, it appears there are also some additional registers/fields which need to be read to determine the detected baud rate as well: `UART_LOWPULSE_MIN_CNT`, `UART_HIGHPULSE_MIN_CNT`, `UART_POSEDGE_MIN_CNT`, and `UART_NEGEDGE_MIN_CNT`.

I guess the configuration would also need some validation, e.g. specifying an explicit baud rate and enabling autobaud is probably not something that should be allowed.

bugadani, 2024-12-19:
> e.g. specifying an explicit baud rate and enabling autobaud is probably not something that should be allowed.

We can solve this by turning the baud rate config into an enum with `Auto` and `Fixed` variants.

bugadani, 2025-01-08:
> We can solve this by turning the baud rate config into an enum with Auto and Fixed variants.

Actually I think this may be wrong and I pretty much shouldn't have said it. Trying to infer from the S3 TRM, it sounds to me like baudrate-detection should be a start/stop function pair that returns a detected baud rate, and not a config option.

We still need a fixed baud rate to transmit and receive data - it's just possible to receive data and measure at the same time, while at the same time, we may or may not receive valid data due to baud rate mismatch.

esp-idf has no built-in users of `uart_ll_set_autobaud_en`, just two test apps that enable/disable it and calculate a baud rate at the end.

bjoernQ, 2025-01-08:
Just had a look at the TRM chapter and after reading it I am not sure this is useful functionality at all. My understanding is that it will measures the minimum low pulse width, the minimum high pulse width etc. and from that it might be possible to calculate a baud-rate.

There is this GitHub issue for ESP-IDF which suggests it's a very "manual" thing: https://github.com/espressif/esp-idf/issues/3336#issuecomment-485410208

I tend to suggest it's something we shouldn't support in the driver.

Dominaezzz, 2025-01-08:
Autobaud is sorta important for writing serial devices without hard coding a frequency.
If you want to implement something like esp-hosted, you want the host to set the baudrate rather than hard code one in the device.

Though it may be manual it's still important I think, maybe as a separate-ish driver. Some kinda autobaud detector struct that can transform into a uart driver once the baud is calculated.

bjoernQ, 2025-01-08:
For sure there might be use-cases for s.th. like that but I guess they are not that common. Especially if the user controls both sides of the communication devices could use a pre-known config and negotiate a switch - s.th. like that

In any case s.th. like a "separate driver" might not be a too bad solution for this

bugadani, 2025-01-09:
I've removed my assignment and I think we can remove the 1.0-blocker label as well.