esp-rs/esp-hal #2649: Direct CPU GPIO access is not supported yet

bjoernQ, 2024-12-02:


bugadani, 2024-12-09:
Just noting here because it's easy to forget and hard to find: This option is, in the C6 TRM, documented under `1.14 Dedicated IO`, in the High-Performance CPU chapter.

ttzytt, 2025-12-29:
Hi, I'm the author of issue #4688. Sorry for creating a duplicate issue. I did search before I posted, but the issue name is slightly different (direct CPU GPIO) from what I expected (dedicated GPIO), so I didn't find this one. 

Anyway, I spent some time trying to understand the implimentation of dedicated GPIO in ESP-IDF, and tried to use this feature in a Rust environment. Eventually, I was able to do so for ESP32-S3 (I currently don't own another model, but I assume there shouldn't be too much change). 

Just posting a possible solution here in case other people need a higher GPIO output frequency (will try input later, but currently my application only needs fast output, so I haven't tried input yet) in Rust. Hopefully, this will be helpful. 

The following code is basically keep toggling pin 4-7. 

```rust
#![no_std]
#![no_main]
#![feature(asm_experimental_arch)]

use core::arch::asm;
use esp_hal::clock::CpuClock;
use esp_hal::gpio::{Level, Output, OutputConfig, OutputSignal};
use esp_hal::timer::timg::TimerGroup;
use embassy_executor::Spawner;
use panic_rtt_target as _;

esp_bootloader_esp_idf::esp_app_desc!();

#[inline(always)]
fn dedic_write_mask(mask: u32, value: u32) {
    // I assume this is architecture-dependent
    unsafe {
        asm!(
            "ee.wr_mask_gpio_out {val}, {msk}",
            val = in(reg) value,
            msk = in(reg) mask,
        );
    }
}

#[esp_rtos::main]
async fn main(_spawner: Spawner) -> ! {
    let config = esp_hal::Config::default().with_cpu_clock(CpuClock::max());
    let peri = esp_hal::init(config);

    let timg0 = TimerGroup::new(peri.TIMG0);
    esp_rtos::start(timg0.timer0);

    // Configure GPIO4..7 as outputs
    let mut p4 = Output::new(peri.GPIO4, Level::Low, OutputConfig::default());
    let mut p5 = Output::new(peri.GPIO5, Level::Low, OutputConfig::default());
    let mut p6 = Output::new(peri.GPIO6, Level::Low, OutputConfig::default());
    let mut p7 = Output::new(peri.GPIO7, Level::Low, OutputConfig::default());

    // Map dedicated CPU GPIO OUT (CPU0) channels 0..3 to GPIO pins 4..7
    OutputSignal::PRO_ALONEGPIO_OUT0.connect_to(&mut p4);
    OutputSignal::PRO_ALONEGPIO_OUT1.connect_to(&mut p5);
    OutputSignal::PRO_ALONEGPIO_OUT2.connect_to(&mut p6);
    OutputSignal::PRO_ALONEGPIO_OUT3.connect_to(&mut p7);

    const MASK: u32 = 0b0000_1111;
    const LO: u32 = 0b0000_0000;
    const HI: u32 = 0b0000_1111;

    loop {
        dedic_write_mask(MASK, HI);
        dedic_write_mask(MASK, LO);
        dedic_write_mask(MASK, HI);
        dedic_write_mask(MASK, LO);
        dedic_write_mask(MASK, HI);
        dedic_write_mask(MASK, LO);
        dedic_write_mask(MASK, HI);
        dedic_write_mask(MASK, LO);
    }
}

[... 10 lines cut ...]
> To install a GPIO bundle, one needs to call [dedic_gpio_new_bundle()](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/dedic_gpio.html#_CPPv421dedic_gpio_new_bundlePK26dedic_gpio_bundle_config_tP26dedic_gpio_bundle_handle_t) to allocate the software resources and connect the dedicated channels to user selected GPIOs. Configurations for a GPIO bundle are covered in [dedic_gpio_bundle_config_t](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/dedic_gpio.html#_CPPv426dedic_gpio_bundle_config_t) structure:

One needs to connect "dedicated channels" to selected GPIOs. So I was browsing [`dedic_gpio_new_bundle` ](https://github.com/espressif/esp-idf/blob/b6e2de03f0e79d02f154e1291fed8e43f300c40d/components/esp_driver_gpio/src/dedic_gpio.c#L203) in `dedic_gpio.c`. I found the function that establishes the connection is `esp_rom_gpio_connect_out_signal`. 

And the same thing can be achieved by something like `OutputSignal::PRO_ALONEGPIO_OUT0.connect_to(&mut p4);` in Rust, which calls [`connect_peripheral_to_output`](https://github.com/esp-rs/esp-hal/blob/9c7c809cec72b11db82d5f72f75d1c075ce654bb/esp-hal/src/gpio/interconnect.rs#L449). 

I found the signals (e.g. `PRO_ALONEGPIO_IN0_IDX`) to connect to the GPIOs in [`dedic_gpio_periph.c`](https://github.com/espressif/esp-idf/blob/b6e2de03f0e79d02f154e1291fed8e43f300c40d/components/esp_hal_gpio/esp32s3/dedic_gpio_periph.c)

Once the connection is established, I just copied the stuff in `dedic_gpio_cpu_ll_write_mask` in ESP-IDF to write to the GPIO. In S3's case, it is: 

```c
__attribute__((always_inline))
static inline void dedic_gpio_cpu_ll_write_mask(uint32_t mask, uint32_t value)
{
    asm volatile("ee.wr_mask_gpio_out %0, %1" : : "r"(value), "r"(mask):);
}
```

Overall, I think the solution will be portable as long as you change the assembly code in `dedic_write_mask` and select the correct channel according to the CPU core you're currently using. 

---

I tried to understand [dedic_gpio_new_bundle](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/dedic_gpio.html#_CPPv421dedic_gpio_new_bundlePK26dedic_gpio_bundle_config_tP26dedic_gpio_bundle_handle_t). Seems like most of the things it does are to manage the dedicated GPIO channels resources in the CPU. It keeps track of the channels that are currently occupied, and finds free channels for the user when registering a new bundle. The channels allocated to the bundle are also dependent on the CPU core on which the function is being called and whether the bundle is input or output. 

I felt like this logic is relatively clear, and I'm interested in helping implementing this feature for esp-hal, please let me know if you have any suggestions for me to get started. Should I keep the interface style simlar to that of ESP-IDF?

bugadani, 2025-12-29:
Thanks for doing the research, I'll draft a PR on friday that is in line with esp-hal guidelines.

ttzytt, 2025-12-30:
BTW @bugadani, I’m interested in contributing if there’s anywhere I can help with the PR you’re drafting to gain experience in open source. As you can see, I’ve already done some research on dedicated GPIO, and I’m pretty free right now since I’m a university student on winter break. Just let me know whenever you have any tasks I can take on.

bugadani, 2026-01-02:
I've opened https://github.com/esp-rs/esp-hal/pull/4699 as an initial implementation. As you can see it's probably a bit involved, that's why I wanted to do this initial work myself. I've left pin bundles out of the implementation, as they may be somewhat niche of a feature, but if you want to give them a try, they might be a good addition to contribute - I have added a comment with a few of the requirements that I managed to think of, and they shouldn't be too hard to implement.

Besides, I'd appreciate any testing, code/documentation review, or examples/doc examples you would do.

Everything is completely optional, of course.

ttzytt, 2026-01-03:
@bugadani  Thanks for drafting the PR. I'll take a look now.

ttzytt, 2026-01-05:
> I agree this may be a bit verbose to use, but it allows you to safely control the pins you want, and not something else by accident.

I agree that this is a reasonable concern.  However, I believe that a better approach would still keep `write_bits()` while adding information in the documentation and adding some runtime checks in debug mode. This is similar to the method you used to deal with the multicore problem of dedicated GPIO, like these: 

```rust
debug_assert_eq!(self.core, Cpu::current());

#[cfg_attr(
    multi_core,
    doc = r#"

<section class="warning">
Note that the driver must only be used on the core that has created it. Do not send the driver to
another core, either directly, or indirectly via a thread that is not pinned to a core.
</section>
"#
)]
```

In debug builds, we could assert that the call does not intend to modify any channels outside the bundle’s mask by comparing against the current output latch:

```rust
#[inline(always)]
pub fn write_bits(&mut self, value: u32) {
    #[cfg(all(debug_assertions, multi_core))]
    debug_assert_eq!(self.core, Cpu::current());

    #[cfg(debug_assertions)]
    {
        let cur = ll::read_out(); // current channel output bits
        debug_assert_eq!(
            (value ^ cur) & !self.mask,
            0,
            "write_bits attempted to change bits outside mask \
             (value=0x{value:08x}, cur=0x{cur:08x}, mask=0x{mask:08x})",
            value = value,
            cur = cur,
            mask = self.mask,
        );
    }

    ll::write(self.mask, value);
}
```

This might come down to design philosophy, but my impression is that esp-hal does emphasize performance. In that spirit, I think it’s useful to offer a more verbose-but-safer API *as long as it doesn’t decrease the performance*.

In this particular case, if the bundle only exposes `set_high/set_low/set_level`, then writing an arbitrary parallel pattern (typical “data bus” use case) would take more instructions compared to a single `write_bits()` call. So I’d still prefer keeping `write_bits()` available for performance-critical patterns, and rely on documentation + debug-only checks to reduce accidental misuse.

--- 

> I think the `Note: Can only control the lowest 4 GPIO channels` makes the instruction-with-immediate approach a bit difficult to use.

Yeah, I think the architectural differences do make the implementation more complicated, and some kind of fallback mechanism seems necessary if we want to use the instruction-with-immediate while keeping the same API.

Alternatively, would it make sense to provide two different bundle APIs? One would use the regular write instruction, where the member `DedicatedGpio`s can be configured dynamically at runtime (and the bundle just recomputes the mask). This API is more flexible but less performant.

The other API would require a compile-time mask and could provide higher performance.

Splitting them into two explicit APIs might also make the behavior easier to reason about - e.g. RISC-V users would know up front that the immediate-based path can only operate on the lowest 4 channels.

One concern I have is how much real performance this actually increases. In my previous testing (in my initial comment), the toggle rate already reached ~120 MHz using the regular instruction (about half the core frequency), which suggests the write itself is already one CPU cycle. You can't possibly be faster than that. 

I’m not sure if this is because the mask never changes (so something gets cached), or because LLVM managed to optimize something away, or if this is simply an ESP32-S3-specific outcome. I guess some more testing and investigation are needed to justify the necessity of including the immediate-based instruction. Ideally, it would be great if we could find some explanation in the official documentation as well.

--- 

If you think the idea is reasonable, I can start implementing an initial version of the bundle API using the regular instructions first. And I can do more investigation on the immediate-based instructions later. 

Would you prefer that I add commits to your current PR or wait until it's merged and then open a new PR?