esp-rs/esp-hal #2660: [RFC] Do we need a prelude?

MabezDev, 2024-12-02:
It's shrunk considerably in the last year or two, maybe we can get away without having one at all?

bjoernQ, 2024-12-02:
I'd say let's remove it - the only "ugly" things are extension traits like `fugit::RateExtU32`

jessebraham, 2024-12-03:
I generally agree that it's not necessarily required, but forcing users to add `fugit` as a dependency to every project using the HAL is also  going to be kind of a pain.

bjoernQ, 2024-12-04:
Even if it's not in the prelude we could still re-export Fugit's extension traits (when it's 1.x)

MabezDev, 2024-12-17:
Given that we mind end up removing `fugit` if it's not stable in time, and based on the discussion here, I think we should try to remove the prelude.

PR #2845 (2024-12-20): Remove prelude
Closes #2660

skip-changelog because other crates were prelude users.
closed 2024-12-20 by bjoernQ