esp-rs/esp-hal #2663: Stabilizing `CpuClock`

MabezDev, 2024-12-02:
As part of https://github.com/esp-rs/esp-hal/issues/2499

We're discussing how to stabilize the minimum amount. Or whether this should be stabilized at all.

### Stabilize `default` and `max`

For each chip we only stabilize the lowest and highest clock speed, and expose `default` and `max` methods.

### Stabilize `CpuClock` as is?

Whilst this enum covers every chip we _currently_ support, could run into an issue with future chips, mostly around dynamic frequencies - if a future chip supports that. Given that it's a new chip, we can introduce a different API for this, so I don't see this being an issue.

### Don't stabilize it

This option is quite self explanatory, we just mark it as `#[unstable]`, and in `esp_hal::Config` and come back to stabilize this at a later date. Clock setting is only really required for use with `esp-wifi`, which itself is currently unstable.

bjoernQ, 2024-12-03:
I vote for `Stabilize CpuClock as is`

MabezDev, 2025-01-07:
Let's stabilize this as is. We can mark it as `#[non_exaustive]` and relax it later on, as we're doing with the clock source enums.

closed 2025-01-07 by jessebraham