esp-rs/esp-hal #2671: Incorrect RESERVED_INTERRUPTS list for PLIC?

wisp3rwind, 2024-12-04:
I'm just browsing the code to understand interrupt handling in `esp-hal` better, and the following line looks off to me:

https://github.com/esp-rs/esp-hal/blob/a189eff51751e7cfc162b6cdd5887fb373b9679f/esp-hal/src/interrupt/riscv.rs#L172

Shouldn't it be

```rust
pub static RESERVED_INTERRUPTS: &[usize] = PRIORITY_TO_INTERRUPT;
```

such that it contains interrupt numbers, not their priorities?

MabezDev, 2024-12-04:
Thanks, I think this is indeed incorrect. It works in the `classic` module because they are the same, but for `plic` and beyond it's wrong.

We should change this to be `PRIORITY_TO_INTERRUPT`.

Dominaezzz, 2024-12-04:
I'm curious if that was the problem on the P4, with the interrupts

closed 2024-12-04 by playfulFence