esp-rs/esp-hal #2687: SPI: one of the half-duplex modes is not supported

bugadani, 2024-12-05:


bugadani, 2024-12-05:
That being said, the S2 TRM explicitly says as an example, that this is supported:

[image]

😩

bjoernQ, 2025-01-07:
I think this boils down to just have a way to enable SIO

Easiest might be to just add a config option for that

bugadani, 2025-01-07:
What would be the effect of that? Every half-duplex transfer would now be 3-wire, regardless of DataMode?

bjoernQ, 2025-01-07:
Yes configuring it per transfer was the other option I thought about.

We shouldn't allow mixing any other DataMode with 3-wire mode in one transfer, then. Since those functions are fallible already, we can probably just check it and return a new error variant if needed

bjoernQ, 2025-01-09:
I guess we reached consent here? (i.e. add a data mode for three-wire SPI)

If yes, I can assign this to me and remove the "investigation" label

PR #2919 (2025-01-17): Support 3-wire-SPI
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [x] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] I have added neces [... cut]
closed 2025-01-17 by MabezDev