esp-rs/esp-hal #2697: GPIO `C-HIDDEN` violations

MabezDev, 2024-12-06:


bjoernQ, 2024-12-20:
I'm not sure about hiding the traits anymore - they might be useful to users as trait bounds?

bjoernQ, 2025-01-10:
we still need to agree if we want the mentioned traits to be hidden or not - if not #2856 should be enough to close this

worst case we can just hide them now and see if someone complains - we can make anything public at any point of time later but not the other way around

bugadani, 2025-01-10:
I don't think we want any of these hidden - although InputPin and OutputPin are implemented for pretty much all pins except on ESP32, users might need to refer to them still if they create pin drivers hidden behind some abstraction.

`GpioRegisterAccess` is hidden right now. Other HALs expose the port of the pins, but that isn't terribly useful for us publicly at the moment. We can mark it as unstable I think, but I'm not sure if it needs to be public at all at least initially.

The rest are unstable so I don't care at the moment.

MabezDev, 2025-01-10:
I believe this is implemented in https://github.com/esp-rs/esp-hal/pull/2926

closed 2025-01-10 by MabezDev