esp-rs/esp-hal #2708: GPIO: The driver's constructor should take the config struct by value, and it should return `Result<Self, ConfigError>`

bugadani, 2024-12-09:


bugadani, 2024-12-19:
Flex should probably be an exception - as it is intended to be able to configured freely with whatever configuration is possible (i.e. it's a lower level driver), we probably don't want any initial configuration, or a single apply_config function for it.

For the rest of the drivers, keeping consistent with the API guidelines makes sense. On ESP32, constructing Output* pins may fail, and we don't currently know how the filter/glitch filter and other, newer features will affect driver configuration.

MabezDev, 2025-01-07:
I don't recall what we needed to investigate, but I think we should do this, and we should do prior to the beta release.

PR #2990 (2025-01-22): Add configuration structs for GPIO drivers, make constructors fallible
Closes #2708
closed 2025-01-22 by MabezDev