esp-rs/esp-hal #2710: GPIO: Prefer compile-time checks over runtime checks where possible, prefer a fallible API over panics.

bugadani, 2024-12-09:


mnmnk43434, 2024-12-09:
Hello, we tried to solve the issue.

This is what we did:

This change addresses the issue where AnyPin operations panic when the underlying pin is incompatible and `listen_with_options` panics with some parameter combinations. We modify the AnyPin implementation to handle incompatible pins gracefully and update `listen_with_options` to avoid panicking.

You can review changes in this commit: https://github.com/mnmnk43434/esp-rs-esp-hal-2710/commit/69fd8f549a51cf93bb1b2c2ad373532f083fb764.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

MabezDev, 2025-01-07:
> AnyPin

I don't think a fallible API makes this situation better, IMO the panic is fine.

> listen_with_options

This only panics when sleeping, so we can make this fallible, and then make `wakeup_enable` fallible. We'll likely want to keep `listen` as infallible though.

MabezDev, 2025-01-07:
With the above in mind, I suppose if we agree, this then stops being a 1.0 blocker.

PR #2916 (2025-01-09): GPIO: wakeup_enable: Return error instead of panic
Closes #2710 I guess
closed 2025-01-09 by MabezDev