esp-rs/esp-hal #2711: GPIO: Avoiding `&mut self` when `&self` is safe to use.

bugadani, 2024-12-09:


MabezDev, 2025-01-07:
I can see how this can be useful in some specific scenarios, but I'm not sure what might happen if a new chip comes out where setting GPIO level is _not_ a single write to a register. That's probably quite unlikely to happen though.

bugadani, 2025-01-07:
FWIW I think this guideline is questionable at best. I don't think implementation details should affect the API.

MabezDev, 2025-01-07:
> FWIW I think this guideline is questionable at best. I don't think implementation details should affect the API.

I somewhat agree, but by that same logic we might want to change `Input::is_high` to `&mut self` instead of `&self` which is currently is. Maybe this isn't so bad though?

bugadani, 2025-01-20:
The pins and pin drivers are not cloneable, so it doesn't matter from a data race standpoint what the API takes. Internally, we'll have to introduce locks anyway around configuration (or at least around RMW operations), most likely one per GPIO bank, but it'll need to be checked a bit more closely.

I recommend sticking with "read shared, write/configure exclusive".

MabezDev, 2025-01-21:
I'd tend to agree, let's table this for tomorrow.

MabezDev, 2025-01-28:
We're sticking with our current signature. Closing.

closed 2025-01-28 by MabezDev