esp-rs/esp-hal #2713: GPIO: If a common Instance trait is used for multiple peripherals, those traits should not have any logic implemented in them.

bugadani, 2024-12-09:


tom-borcin, 2024-12-18:
Hide all pin trait methods

bugadani, 2024-12-19:
The only relevant trait with methods is Pin. Pasting here the rustdoc output, we have these:

```rust
pub trait Pin: Sealed {
    // Required method
    fn number(&self) -> u8;

    // Provided method
    fn degrade(self) -> AnyPin
       where Self: Sized { ... }
}
pub trait InputPin:
    Pin
    + Into<AnyPin>
    + 'static { }
pub trait OutputPin:
    Pin
    + Into<AnyPin>
    + 'static { }
```

I think in terms of public functions, this is reasonable. The other traits are intended to be hidden by #2693

MabezDev, 2024-12-19:
I think the only question is should `number` and `degrade` be inherent impls on each gpio struct, instead of in the trait?

I can see a world where a user uses our traits to be generic over some i/o pins and might want to access something like pin number. It's whether we expose it now, or not. We also have to be careful to not put the kitchen sink in the trait methods over the long term (this guideline should prevent us from doing this, in theory).

bugadani, 2024-12-19:
`degrade` is questionable - `InputPin` and `OutputPin` both implement `Into<AnyPin>`. `fn number` is as close to a GPIO `fn info()` as possible right now, and we don't really publish those currently, so one could argue that we can hide that as well for the sake of consistency.

PR #2926 (2025-01-10): Remove logic from InputPin and OutputPin
I'd say this closes #2713

Pin's `number` and `degrade` remain public. `input_signals` and `output_signals` remain "private". Rest of the traits are unstable so I didn't touch them.

This is close enough to how the embassy_stm32 Pin trait [looks](https://docs.embassy.dev/embassy-stm32/git/stm32f030rc/gpio/trait.Pin.html) so I don't think we should grind up more time on the place of degrade/number.
closed 2025-01-10 by bjoernQ