esp-rs/esp-hal #2716: Avoid repeating the enum name in the enum variants

SergioGasquez, 2024-12-10:
There are several places where we repeat the enum name in each enum variant:
- UART: https://github.com/esp-rs/esp-hal/blob/main/esp-hal/src/uart.rs#L338-L385 
  - Maybe it's fine for `StopBits`
- SPI:  `Command` and `Address` 
  - We will have to come up with a way of having numbers on variant names

bugadani, 2024-12-13:
First, we should probably figure out why clippy doesn't trigger on these cases. https://rust-lang.github.io/rust-clippy/master/index.html#enum_variant_names exists and is a warn-level lint by default.

SergioGasquez, 2024-12-18:
I think `Partity` should be changed to  `None`, `Even`, `Odd`. Then we would need to decide a way to name numerics variants, the most difficult being StopBits (as it has 1.5 variant), here are some options:
- `Stop1`, `Stop1p5`... [`stm32h7xx-hal`](https://github.com/stm32-rs/stm32h7xx-hal/blob/0735f793e6d432653f2a7f1d8a2d05f073ca4f17/src/serial.rs#L101-L110)  (This repeats partially the enum name)
-  `STOP1`, `STOP1P5`.... [`embassy`](https://github.com/embassy-rs/embassy/blob/main/embassy-stm32/src/usart/mod.rs#L117-L126) (This repeats partially the enum name)
- `ONE`, `ONE_POINT_FIVE`... [`uefi-rs`](https://github.com/rust-osdev/uefi-rs/blob/1e5380e774bdad01bdefdf9b267d76bdb5f0cb0d/xtask/src/device_path/spec.rs#L664-L677) 
- [`ONE`, ONE_HALF`](https://github.com/Kalapaja/kampela-firmware/blob/6ca9d1a16cc397048376019fd9142d864230f6e4/kampela_experiments_efm32pg23/efm32pg23_fix/src/eusart1_s/framecfg.rs#L264-L273)
- [`_1`, `_1_5`](https://github.com/Kalapaja/kampela-firmware/blob/6ca9d1a16cc397048376019fd9142d864230f6e4/kampela_experiments_efm32pg23/efm32pg23_fix/src/eusart1_s/framecfg.rs#L264-L273)
   
See more examples [here](https://github.com/search?q=%22enum+StopBits%22+language%3ARust+&type=code) 

I think I would go either with `_number`  or written number, so the "most weird scenario" would be `StopBits::_1_5` or `StopBits::OneHalf`/`StopBits::OnePointFive` and `Address::_28` or `Address:TwentyEigth`

bugadani, 2024-12-18:
Turns out, clippy can be configured and has `avoid-breaking-exported-api: Suppress lints whenever the suggested change would cause breakage for other crates.`. We may want to disable this while we are allowed to break API :)

bjoernQ, 2024-12-19:
One more variant: `StopBits::One`, `StopBits::OneHalf`, `StopBits::Two`

Something like `StopBits::_1P5` really hurts my eyes. The all-upper-case naming violates the naming convention.

bugadani, 2024-12-19:
> One more variant: StopBits::One, StopBits::OneHalf, StopBits::Two

That's a good one!

jessebraham, 2024-12-19:
> One more variant: `StopBits::One`, `StopBits::OneHalf`, `StopBits::Two`

Should be `StopBits::OneAndAHalf` or something in this case, as `OneHalf` is 0.5, so slightly less nice 😅

MabezDev, 2024-12-19:
I quite like what we did for drive strength: https://docs.esp-rs.org/esp-hal/esp-hal/0.22.0/esp32c6/esp_hal/gpio/enum.DriveStrength.html, prefixing it with `I` for current (presumably, I don't actually know :sweat_smile:). Is there an appropriate prefix we could add here? Maybe `StopBits::B1P5`?

SergioGasquez, 2025-01-07:
https://github.com/esp-rs/esp-hal/pull/2893 resolves some of the naming issues

jessebraham, 2025-01-07:
I've found a handful of other enums which violate the naming rules as well, which I will get a PR put together for either today or tomorrow.

MabezDev, 2025-01-08:
What's left to do here?

Removing the investigation label, just because we have settled on the `_` variant prefix.

bugadani, 2025-01-08:
> What's left to do here?

We should codify this choice in the guidelines.

SergioGasquez, 2025-01-08:
> What's left to do here?

Also, we should change the naming accordingly in SPI `Address` and `Command`  enums.

jessebraham, 2025-01-08:
> Also, we should change the naming accordingly in SPI `Address` and `Command` enums.

Will have a PR for this shortly

PR #2906 (2025-01-08): Fix naming violations for SPI `Address` and `Command` enums
There are still a handful of enums which violate this rule, but they are not in drivers which are being stabilized, so I guess at least within the scope of the issue this closes #2716?
closed 2025-01-08 by jessebraham