esp-rs/esp-hal #2720: Re-use WiFi/BLE calibration result

maufl, 2024-12-10:
## Motivations
I'd like to save as much power as possible. My use case is a device that spends most time in deep sleep and periodically wakes up to send a BLE advertisement. 
As far as I know, right now when we initialize the WiFi/BLE stack we always do a full calibration.

## Solution
We could maybe reuse the calibration result from a previous initialization.

## Additional context
This came up as a possible optimization in the Matrix channel. The project I want this for can be found here https://github.com/maufl/sensor2

bjoernQ, 2025-01-06:
We probably should support all calibration modes
```c
#[doc = "< Do part of RF calibration. This should be used after power-on reset."]
pub const esp_phy_calibration_mode_t_PHY_RF_CAL_PARTIAL: esp_phy_calibration_mode_t = 0;
#[doc = "< Don't do any RF calibration. This mode is only suggested to be used after deep sleep reset."]
pub const esp_phy_calibration_mode_t_PHY_RF_CAL_NONE: esp_phy_calibration_mode_t = 1;
#[doc = "< Do full RF calibration. Produces best results, but also consumes a lot of time and current. Suggested to be used once."]
pub const esp_phy_calibration_mode_t_PHY_RF_CAL_FULL: esp_phy_calibration_mode_t = 2;
#[doc = " @brief PHY calibration mode\n"]
```

MabezDev, 2025-05-02:
We now support other calibration modes: https://github.com/esp-rs/esp-hal/pull/3383 but still lack the ability to reuse stored calibration from NVS.

maufl, 2025-05-04:
Thanks really cool, thanks! I was wondering whether, instead of in NVS, we could put the initialization data in RTC RAM using https://docs.espressif.com/projects/rust/esp-hal/1.0.0-beta.0/esp32c6/esp_hal/attr.ram.html

bjoernQ, 2025-05-05:
With  ESP_WIFI_CONFIG_PHY_SKIP_CALIBRATION_AFTER_DEEP_SLEEP set to `true` you won't even need to keep the calibration data in RTC/LP RAM

PR #4001 (2025-08-28): Add a way to get/set calibration data
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] [... cut]