esp-rs/esp-hal #2725: `fill_fifo` seems redundant vs `read_bytes`

MabezDev, 2024-12-10:


bugadani, 2024-12-13:
What is `fill_fifo`? I can't find it in `uart.rs` at all.

MabezDev, 2024-12-16:
> What is `fill_fifo`? I can't find it in `uart.rs` at all.

Sorry, it's `drain_fifo`

bugadani, 2024-12-16:
`read_bytes` waits for the buffer to fill up, but I agree we probably don't need two of these.

bjoernQ, 2024-12-19:
> `read_bytes` waits for the buffer to fill up, but I agree we probably don't need two of these. We should probably go with the non-blocking semantics of dran_fifo and let the user handle the waiting.

I agree with the exception that `drain_fifo` is a weird name in public API - something like `try_read`? (or maybe not - I'm bad at naming but I can spot weird naming when I see it 😆 )

PR #2895 (2025-01-07): UART: Remove blocking version of `read_bytes` and rename `drain_fifo` to `read_bytes` instead
Hopefully I have understood the issue correctly 😅 But should close #2725. If there are any changes that need to be made let me know.
closed 2025-01-07 by jessebraham