esp-rs/esp-hal #2732: `AtCmdConfig` should be a part of `Config`

MabezDev, 2024-12-10:


bugadani, 2024-12-12:
I'm not 100% sure this should be. A separate AtCmdConfig struct allows us to create APIs like `async fn wait_for_at_command(&mut self, config: &AtCmdConfig)`, which would be very annoying to do with the full Config struct.

MabezDev, 2025-01-06:
Hmm, for async I agree the above API is nicer. For blocking + interrupts, the path seems less clear. The easy option a `Blocking` specific `set_at_config` method (and one to clear it?).

If it were part of the global uart config, the above API you suggested could still exist, it would just be `async fn wait_for_at_command(&mut self)`, where the config is drawn from the underlying config. This would however, involve storing the config in the driver struct.

MabezDev, 2025-01-06:
Just pointing out we could punt this to post 1.0, if there isn't a clear path here just yet. We should still try to resolve this and create a path if possible though.