esp-rs/esp-hal #2735: ClockSource: should this be non-exhaustive?

MabezDev, 2024-12-10:


MabezDev, 2024-12-10:
Provided we support all sources for the chips we have now we can leave it as complete, we can add new sources (and cfg away old ones) when (and only when) adding new chips, e.g esp32s6 might have some new clock source

jessebraham, 2024-12-19:
On the one hand: I think it's better to be conservative in this regard and mark enums `#[non_exhaustive]` if we're not sure. This can be removed at any time in a non-breaking way AFAIK, so there is not much harm in including it IMO.

However, in this case... it's not like existing chips will suddenly gain new clock sources, so as long as we're certain that everything is present then yes I agree, we can still add new sources for future devices in a non-breaking way using `cfg`.

bjoernQ, 2024-12-19:
We might cover all sources but I'm undecided because making it non-exhaustive wouldn't harm since user-code would probably not match on a clock-source anywhere and if they do - it's not that awful

jessebraham, 2024-12-19:
> We might cover all sources but I'm undecided because making it non-exhaustive wouldn't harm since user-code would probably not match on a clock-source anywhere and if they do - it's not that awful

Even if they do have a match (can't imagine why somebody would), removing the annotation would just result in an unreachable match arm warning for the catch-all arm which would need to be included previously, there wouldn't be an error I don't think.

MabezDev, 2025-01-07:
Let's be safe and mark them as `#[non_exaustive]` then. As @jessebraham we can relax this in a non-breaking way later down the line.

Whoever takes this should also update the migration guide to reflect the decision here.

PR #2912 (2025-01-08): Mark ClockSources as #[non_exhaustive]
Closes #2735
closed 2025-01-08 by jessebraham