esp-rs/esp-hal #2736: [RFC] MSRV policy going forward

MabezDev, 2024-12-11:
With #2615 we ourselves in a situation where we have to bump the MSRV even though we're not using any new compiler features on the RISCV side. This is likely to happen again in the future.

The simplest proposal is to maintain two MSRVs, essentially a "arch-specific" MSRV. This will allow us to keep the RISCV one lower, and allow us to bump the Xtensa MSRV when we run into compiler issues.

bugadani, 2024-12-11:
We can also specify "latest stable" globally and accept that while a patch release may keep API compatibility, an update may break a build that uses an older compiler. It's semver-compatible, though maybe not the most polite way to approach the issue.

jessebraham, 2024-12-11:
> The simplest proposal is to maintain two MSRVs, essentially a "arch-specific" MSRV. This will allow us to keep the RISCV one lower, and allow us to bump the Xtensa MSRV when we run into compiler issues.

Not sure I agree with this, this will complicate our CI and we will not be able to specify `rust-version` in the Cargo manifest anymore (which maybe is not the end of the world, but does provide a nicer experience when using unsupported toolchains IMO).

jessebraham, 2024-12-11:
> We can also specify "latest stable" globally and accept that while a patch release may keep API compatibility, an update may break a build that uses an older compiler. It's semver-compatible, though maybe not the most polite way to approach the issue.

This is probably a reasonable approach, but will have to think about it.

bjoernQ, 2024-12-11:
One advantage of bumping both MSRVs this time would be that we get `core::error::Error` - implementing it on our error types would be a good thing

Other than that, it's probably not a problem to have "arch specific" MSRVs in general

jessebraham, 2024-12-17:
I'm sort of leaning towards `stable` or `stable - 1` at this point, just to make things a bit easier on ourselves with regards to potential future compiler issues. We can always extend the MSRV in the future, and I don't think this is an unreasonable limitation. There is of course always the caveat that things _may_ work with older compiler versions, we will just choose not to officially support them. Curious to hear what others think about this, though.

closed 2025-01-14 by jessebraham