esp-rs/esp-hal #2740: ADC increases current consumption during deep sleep

dimpolo, 2024-12-11:
## Bug description

Deep sleep current consumption is increased by 250 µA on my ESP32-S3 if ADC was enabled before entering deep sleep.

## To Reproduce

1. run `Adc::new`
2. run `Rtc::sleep_deep`
3. Observe increased current consumption

I added the following before entering sleep which solved my issue.

```rust
let sensors = unsafe { &*esp32s3::SENS::ptr() };
sensors.sar_power_xpd_sar().modify(|_, w| unsafe { w.force_xpd_sar().bits(0) });
```

## Expected behavior

Not sure. Maybe calling sleep should disable the ADC, maybe providing a `Adc::disable` method and documentation would be enough

## Environment

- Target device: ESP32-S3
- Crate name and version: esp-hal 0.22.0

MabezDev, 2024-12-11:
I think we have a lot of work to do in this area, both with regards to the ADC API & low power.

Exposing some kind of unsafe `Adc::disable` API might be a good first step for now. I think it would be a good idea to look at how ESP-IDF handles this too, what kind of checks to they run before sleeping and whether they prevent sleep whilst certain domains are still active.

dimpolo, 2024-12-11:
If `Adc::disable` takes `self` by value it wouldn't have to be unsafe

closed 2026-11-13 by MabezDev