esp-rs/esp-hal #2743: UART: Unexpected results when applying a config while data is being clocked out

bjoernQ, 2024-12-12:
Tested with esp-hal 0.22.0 on S3 (doesn't seem to be chip specific)

Given this code 
```rust
let mut lin_uart = esp_hal::uart::Uart::new_with_config(
        peripherals.UART1,
        esp_hal::uart::Config::default().baudrate(9600),
        peripherals.GPIO4,
        peripherals.GPIO5,
    )
    .unwrap();

    loop {
        lin_uart
            .apply_config(&esp_hal::uart::Config::default().baudrate(9600))
            .unwrap();

        lin_uart.write_byte(0xaa).unwrap();
        nb::block!(lin_uart.flush_tx()).unwrap();

        // d.delay_micros(1000); // !!!!!!

        lin_uart
            .apply_config(&esp_hal::uart::Config::default().baudrate(19200))
            .unwrap();

        lin_uart.write_bytes(&[0x55, 0x01, 0x02, 0x03]).unwrap();
        nb::block!(lin_uart.flush_tx()).unwrap();

        d.delay_millis(5);
    }
```

Clocking out 0xaa happens at 19200 baud:

[image]

Uncommenting the delay shows the expected bit width:

[image]

bjoernQ, 2024-12-12:
Update: on ESP32 things look even more weird as it seems

bugadani, 2024-12-12:
Can you please clarify what exactly we see on the screenshots?
[image]

bjoernQ, 2024-12-12:
Here using 9600 / 115200

w/o the delay

[image]

bit-width is  +8.68 which matches 115200 - not 9600

with delay

[image]

bit-width is ~104 which matches 9600

bugadani, 2024-12-12:
This issue has an interesting consequence:
- Currently we allow splitting a UART into RX/TX halves.
- Those halves may be controlled by different tasks.
- Both halves configure the whole peripheral using the same Config struct.
- `->` Configuring the RX half for example may interfere with an ongoing TX transmission.
- In the future we may support DMA even with the split halves, further complicating the problem as the CPU won't always be in control.

bugadani, 2025-02-19:
What baffles me in this issue is that the TX FIFO was flushed before applying the configuration. Do we have a bug there, where `flush` doesn't actually wait for the FIFO to be drained?

Either way, should we flush in `apply_config`, or should we instead return a Busy error so that we don't block the user accidentally, but ensure that there is no data being clocked out?

bjoernQ, 2025-02-19:
I guess blocking until the FIFO is drained is a bit more "in-line" with other APIs - but changing the configuration is a different enough use-case to justify a different behavior - would users actually do anything else than re-trying in this case? With async they might yield and let other tasks do some work

MabezDev, 2025-02-28:
Closed via #3151

closed 2025-02-28 by MabezDev