esp-rs/esp-hal #2748: Clean up remaining raw register manipulations

bugadani, 2024-12-12:
We can collect such cases here and tick them off as we go:

- https://github.com/esp-rs/esp-hal/blob/d66281609b4c593ad07d0eb68a86efbbdb364f23/esp-hal/src/system.rs#L584-L585
- https://github.com/esp-rs/esp-hal/blob/d66281609b4c593ad07d0eb68a86efbbdb364f23/esp-hal/src/system.rs#L519-L520
- https://github.com/esp-rs/esp-hal/blob/d66281609b4c593ad07d0eb68a86efbbdb364f23/esp-hal/src/system.rs#L326
- https://github.com/esp-rs/esp-hal/blob/d66281609b4c593ad07d0eb68a86efbbdb364f23/esp-hal/src/system.rs#L375

bjoernQ, 2025-08-25:
Add MMU related registers to the PACs and use them - see https://github.com/esp-rs/esp-hal/pull/3979#discussion_r2297450537

MabezDev, 2026-01-07:
I believe this is now resolved?