esp-rs/esp-hal #2752: UART: transmitting BRK is not supported

bugadani, 2024-12-12:


zpg6, 2024-12-12:
ESP-IDF offers a good starting point:

```
int uart_write_bytes_with_break(uart_port_t uart_num, const void *src, size_t size, int brk_len)
```

But this only allows a BRK after the bytes. I am interested in using this for LIN bus - so a break before bytes is needed.

To keep it simple, I would love to see the function for this just be:

```rust
fn send_break(bit_length: u8) -> nb::Result<(), Error> { }
```

Eventually, async would be great as well.

zpg6, 2024-12-21:
For LIN - inverting the level of the TX line does a nice job. Accomplished this by using the PAC to set `UART_TXD_INV` to `1` for the break duration, then `UART_TXD_INV` back to `0`.

MabezDev, 2025-01-08:
We can add this after 1.0 in a non-breaking way. Looks like it should be a new API to send breaks, and @zpg6 has some draft PRs proving this.