esp-rs/esp-hal #2755: `ok()` mistakenly used for error handling

ede1998, 2024-12-12:
Thanks for maintaining and developing ESP HAL. It was really fun when I developed my previous project with it.

## Bug description

I just stumbled over some example code which looked weird to me:

https://github.com/esp-rs/esp-hal/blob/3a03dd88c774b68533f7c539c7480d19f05fc9ec/examples/src/bin/wifi_embassy_dhcp.rs#L96

`embassy` returns a `Result` here on which `.ok()` is called.
This looks like the intention was to check whether the spawing succeeded. However, this does something else:
It turns the `Result` into an `Option<()>` that has `Some(())` on success and `None` on error.
It just silently ignores the return value basically.

The idiomatic Rust way would be `unwrap()` or `expect("What")` instead of `ok()` to trigger a panic on error.
Alternatively, you can ignore the result idiomatically via `let _ = spawner.spawn(connection(controller));`

What worries me is that this code pattern seems to be quite common in this repo and might trap beginners that just assume it handles the error. Github Search finds 30 matches:
https://github.com/search?q=repo%3Aesp-rs%2Fesp-hal%20%20%22.ok()%3B%22&type=code

bugadani, 2024-12-12:
Thanks for noticing. You are right in that we should be more disciplined in our examples.

Just a side note and a bit of fun fact, for your example you picked one of the cases where we can safely ignore any errors :) Each embassy task by default comes with a pool (the number of running instances you can have at the same time for a particular task) of 1. Since we are spawning a single instance of each task, the operation cannot fail.

ede1998, 2024-12-12:
Thanks for the quick reaction and the trivia.

Just FYI: Unfortunately it does not seems to be restricted to examples when I look through the search hits.

bjoernQ, 2024-12-13:
Besides in examples we _should_ only use `.ok()` if we definitely know a function cannot fail given the input _or_ if we really don't care if a function fails.

If you see any usage of `.ok()` where this might not be the case, please let us know

MabezDev, 2025-07-18:
We'll be reworking the examples. I've relabelled the issue to target the `ok()` used (probably incorrectly) inside esp-wifi and esp-ieee802154 instead.