esp-rs/esp-hal #2770: I2C: Generally, follow common "good practices" and idiomatic Rust style

bugadani, 2024-12-13:


tom-borcin, 2024-12-18:
Investigate if FIXME and TODO are user facing

bjoernQ, 2024-12-19:
## FIXME

In `Driver:::setup`

```rust
        // Configure filter
        // FIXME if we ever change this we need to adapt `set_frequency` for ESP32
        set_filter(self.register_block(), Some(7), Some(7));
```

In `set_frequency` (ESP32)
```rust
        // FIXME since we always set the filter threshold to 7 we don't need conditional
        // code here once that changes we need the conditional code here
        scl_high -= 7 + 6;
``` 

```rust
    #[cfg(not(any(esp32, esp32s2)))]
    /// Reads all bytes from the RX FIFO.
    fn read_all_from_fifo_blocking(&self, buffer: &mut [u8]) -> Result<(), Error> {
        // Read bytes from FIFO
        // FIXME: Handle case where less data has been provided by the slave than
        // requested? Or is this prevented from a protocol perspective?
        for byte in buffer.iter_mut() {
...
```

```rust
#[cfg(any(esp32, esp32s2))]
    /// Reads all bytes from the RX FIFO.
    fn read_all_from_fifo_blocking(&self, buffer: &mut [u8]) -> Result<(), Error> {
...
        // Read bytes from FIFO
        // FIXME: Handle case where less data has been provided by the slave than
        // requested? Or is this prevented from a protocol perspective?
        for byte in buffer.iter_mut() {
...
```

## TODO

In `config`
```rust
    // TODO: when supporting interrupts, document that SCL = high also triggers an interrupt.
    pub timeout: Option<u32>,
```

```rust
    // TODO: missing interrupt APIs
```

MabezDev, 2025-01-10:
I don't think these are user facing issues worth blocking on. It seems timeout will be resolved by https://github.com/esp-rs/esp-hal/pull/2864

MabezDev, 2025-10-15:
I just looked at the driver and the remaining FIXME's are related to an esp32 workaround. Closing.

closed 2025-10-15 by MabezDev