esp-rs/esp-hal #2778: Missing https://docs.rs/embedded-hal/1.0.0/embedded_hal/spi/trait.SpiDevice.html impl

bjoernQ, 2024-12-13:


jessebraham, 2024-12-20:
Adding a new third-party trait implementation shouldn't be a breaking change, should it? Don't recall why we decided to implement this, it should be additive functionality AFAIK. Am I missing/forgetting something?

bugadani, 2024-12-20:
I think this is in "investigation" because I'm stubbornly keep repeating that the SPI bus drivers we have should not implement SpiDevice anyway, and that deserved the investigation label instead of a 10-minute detour.

bugadani, 2024-12-20:
FWIW the resolution of this issue is still just adding more code so the 1.0-blocker can probably be removed.

MabezDev, 2025-01-08:
Agree, we can add this post 1.0 as it will be a completely new set of structs.