esp-rs/esp-hal #2784: C-INTERMEDIATE

bugadani, 2024-12-13:


MabezDev, 2025-01-07:
> Should write, read, write_read return the number of transmitted/received bytes on error?

I'm not sure it's that useful. The `read` & `write` API will read/write until the buffer is full/empty - unless an error occurs, in which case the buffer state is unknown (and I'm not sure a partial filled buffer is even useful).

> Should transaction return the number of successfully completed operations?

This seems more useful, and might allow the user to continue the transaction in some cases if the user know where the transaction failed - it might also be helpful for debugging.

MabezDev, 2025-01-10:
Closing this in favour of the general one: https://github.com/esp-rs/esp-hal/issues/2820, it expands a bit more on the idea above.

closed 2025-01-10 by MabezDev