esp-rs/esp-hal #2785: C-GENERIC: `write`, `read`, `write_read` should accept `impl AsRef<[u8]>` and `AsMut`

bugadani, 2024-12-13:


MabezDev, 2025-01-10:
IIRC, this is bad for code size. If you have a driver function that you could pass a `Box<[u8]>` or `&'static mut [u8]` into, you'll get copy of the functions for each type passed in. In the interest of code size, which is much more important on our platforms I believe we should make the caller handle the conversion.

If we all agree, I believe to close this we should update the API guidelines to note this as another exception.

bugadani, 2025-01-10:
Code bloat could I believe largely be mitigated with the following pattern:

```rust
fn outer(&mut self, data: impl AsRef<[u8]>) {
    fn inner(this: &mut Self, data: &[u8]) { ... }

    inner(self, data.as_ref()
}
```

Dominaezzz, 2025-01-14:
> (I think technically a &T -> AsRef change would count as non-breaking?)

Does breaking inference count as a breaking change?
(I'm trying to recall if this changes how inference works)

MabezDev, 2025-01-28:
A somewhat related discussion about how to accept `Config`'s here: https://github.com/esp-rs/esp-hal/pull/3011#discussion_r1932556885.

(removing i2c label, as I think this applicable to most drivers)

MabezDev, 2025-04-24:
We had some more discussion in the impl PR: https://github.com/esp-rs/esp-hal/pull/3415. I also found a snippet right at the bottom of the [`AsMut` docs](https://doc.rust-lang.org/std/convert/trait.AsMut.html#examples): 

> Note, however, that APIs don’t need to be generic. In many cases taking a &mut [u8] or &mut Vec<u8>, for example, is the better choice (callers need to pass the correct type then).

This is also discussed a bit further in the "guidelines": https://rust-lang.github.io/api-guidelines/flexibility.html#c-generic. This guideline is a bit weird, because instead of describing it as a guideline, it delves into a pro/con making this (imo) too subjective to be a guideline.

There are a number of other points too. The fact that `embedded-hal` doesn't respect this guideline is probably an indication we shouldn't either. The playground example from @Dominaezzz (see PR comments) also shows that inference is worse with the added layer of `AsRef` indirection.

At this point I am now convinced we _shouldn't_ do this, at least not on the main API entry points. We may expand the API surface to support a more generic interface, but this should be in _addition_ to the dead simple slice based APIs.

Closing.

closed 2025-04-24 by MabezDev