esp-rs/esp-hal #2797: Expose the  option (This is important for chunked transfers in a single transaction)

bjoernQ, 2024-12-13:


MabezDev, 2025-01-10:
For the same reasons as https://github.com/esp-rs/esp-hal/issues/2795, this can be implemented later.

Dominaezzz, 2025-01-10:
Unlike #2795 I don't think this can come later. This is not the sort of setting one puts in a config, it's the sort of thing you'd specify per-transfer. If there was some kind of TransferConfig struct that this could go into, then yes, it could be postponed, but right now it can't.

I think this needs a more concrete plan

Dominaezzz, 2025-01-13:
> Regardless, this is strictly to do with segmented transfers, right?

No, this applies to CPU transfers as well. It's used quite a bit in esp-idf-hal.

> There will be something like this, probably `DeviceConfig` that could contain this, which would come when we add our own Hardware managed `SpiDevice` later on.

DeviceConfig sounds like something you would set once per device driver created, this option however is something that would still be specified per transfer, which I think should be a separate config.

bugadani, 2025-01-13:
We might go in the same direction as with I2C transfers, and provide a low-level transaction api to build stuff like this. Regardless, it's not needed right now for what we intend to stabilize.

Dominaezzz, 2025-01-13:
> We might go in the same direction as with I2C transfers

That would be a little too restrictive 😅

I suppose it doesn't have to be stabilised now, my concern is I don't think you can add this afterwards, in a nice way.

bugadani, 2025-01-13:
How do you know it would be restrictive? We haven't even _started_ to design it. Anyhow, PRs are welcome in case you'd like the functionality before we get to it.

MabezDev, 2025-01-21:
> Almost every spi transfer method would get an extra keep_cs_active: bool parameter, and I'm not sure that would get

but this would all still be internal right? and we can expose it via the low level transaction builder API in the future.

> I'd make a PR but y'all still need to decide what to stabilise.

Doesn't even have to be a PR, just some more concrete use cases and scenarios might help me understand a bit more about what needs to be done here. I've seen the impl in esp-idf-hal, I don't think it's that bad, in each transaction loop it checks if there is more data to send and if there is it sets the keep cs active flag.

Dominaezzz, 2025-01-21:
> but this would all still be internal right? and we can expose it via the low level transaction builder API in the future.

This would have to be external really, like esp-idf does it.

A "transaction builder" API makes sense for something like I2C where the hardware interface makes you prepare a list of commands in some registers.
It doesn't make sense for SPI as each transfer is sent by itself. If this were for configurable segmented transfers, a transaction builder API could make sense.

If we're still trying to expose an inherent API, this flag has to be exposed in the lower level (no chunking) APIs.

> Doesn't even have to be a PR, just some more concrete use cases and scenarios might help me understand a bit more about what needs to be done here.

That works, I'll come up with scenarios.

Dominaezzz, 2025-01-29:
Use cases for a public "keep cs active" option.

- https://github.com/rust-embedded/embedded-hal/issues/583
- https://github.com/rust-embedded/embedded-hal/issues/478

I think these two issues actually describe the two use cases I'm thinking of pretty well.

The first being you may not know when the transaction is going to end.
The second being you may want to do something mid transaction like toggle a out of band GPIO pin.

Neither of which can be achieved with a transaction builder API but the hardware itself can certainly do it.

bugadani, 2025-07-16:
>  the SPI hardware lets you select multiple CS pins at once

Yes, and this is very dangerous because you can get two devices drive the bus actively in different directions. This is probably not something esp-hal should let you do, but you'd have the PAC and software CS to commit crimes.

> The second one feels dangerously stateful

And the same issue exists with `transfer_with_config`, too, you just can't deassert the chip select without starting another transfer. If your bus is shared, you can happily start transfers one after the other with different chip selects and different `KEEP_CS_ACTIVE` settings, and the end result would probably be up to sheer luck.

KEEP_CS_ACTIVE is a difficult problem to combine with shared bus anyway. But a shared bus is already a difficult problem in the context of esp-hal because we can't just assume the user to be content with the way we'd implement sharing, and there is no single Mutex trait currently available that we could use to inject a lock.

Bus locking will be a funny problem to solve.