esp-rs/esp-hal #2807: Should we get rid of nb::Result for inherent functions?

bjoernQ, 2024-12-13:


jessebraham, 2024-12-17:
I think probably it's reasonable to return an `Option` from `read_byte` based on whether or not we're able to read, then we can just converted this into the appropriate return type in the `embedded_hal_nb` implementation (e.g. `Some` -> `Ok`, `None` -> `Err`).

bjoernQ, 2024-12-18:
Had another look and while for something like `read_byte` returning `Option` even feels more natural there are other places where it's not that easy.

e.g. `write_byte`,`flush_tx`, some functions in the crypto-accelerator drivers (some are questionable - e.g. `Hmac::configure` never returns `WouldBlock`), there are usages in SPI and TWAI, in `PeriodicTimer::wait` and `ADC`

closed 2025-01-17 by MabezDev