esp-rs/esp-hal #2811: C-QUESTION-MARK: Examples use ?, not try!, not unwrap

bugadani, 2024-12-13:


playfulFence, 2025-01-10:
Last night I wanted to end the day with this issue as a quick win, but something went wrong....
[image]

Turns out, using `?` is not so easy in the context of our drivers - we simply don't return an `error` from `main`. 
@bugadani tried to make a solution for this (thanks)((it's shown in the screenshot below))
[image]

It seems to me that at this point there's not much point in fighting over the use of `? ` in our case. 
1) The solution to this problem for the documentation is already very cursed by itself
2) `?` is not applicable in the context of the main function in our driver, so the doc example will make no sense. 
3) This Rust API guideline uses “should” wording, which means that it's not a strict prescription. 

With everything said above, I would prefer to close the `C-QUESTION-MARK` related issues. I look forward to your opinions!

bugadani, 2025-01-10:
> ? is not applicable in the context of the main function in our driver, so the doc example will make no sense.

I think this is the deciding factor. The `fn main() -> Result<(), Box<dyn Error>>` pattern doesn't apply to us, and the user has to work pretty hard if they want it. If someone can come up with a solution like in the screenshot, they can also replace unwraps in an example.

playfulFence, 2025-01-10:
Closing this

closed 2025-01-10 by playfulFence