esp-rs/esp-hal #2814: We do not distinguish NACK sources (address/data) which embedded-hal supports. Adding this would change the Error type, most likely renaming the `AckCheckFailed` variant.

bugadani, 2024-12-13:


bjoernQ, 2024-12-17:
Given there is only NACK_INT I wonder how to best check this - check the FIFO_COUNT?

If that works should be distinguish between NACKed writes / reads?

Maybe also related to #2820 (and #2784) since we could report more information with the error

bjoernQ, 2024-12-19:
We could also change the variant to `NoAcknowledge(NoAcknowledgeSource)` (like in `embedded-hal`) - we can for now just always use `Unknown` but will be able to figure out more fine-grained errors in future - we can even only have the `Unknown` variant for now since that enum will be non-exhaustive

bjoernQ, 2024-12-19:
But this still wouldn't allow us to augment the error with more information in future - not sure if (especially in this case) there would be a real benefit to the user but maybe in other cases - which is more something into the direction of #2833

closed 2025-01-10 by jessebraham