esp-rs/esp-hal #2816: Memory leak when writing to BleConnector

s8alprys, 2024-12-13:
## Bug description

Since version 0.11.0 of esp-wifi, the `BleConnector` leaks memory when calling `write()` or `write_all()`. This does not happen in version 0.10.1.

## To Reproduce

### main.rs

```
#![no_std]
#![no_main]

use embassy_executor::Spawner;
use embassy_time::{Duration, Timer};
use embedded_io::Write;

use esp_backtrace as _;
use esp_wifi::{ble::controller::BleConnector};

#[macro_use]
extern crate alloc;
use core::mem::MaybeUninit;

use esp_alloc as _;

fn init_heap() {
    const HEAP_SIZE: usize = 64 * 1024;
    static mut HEAP: MaybeUninit<[u8; HEAP_SIZE]> = MaybeUninit::uninit();

    unsafe {
        esp_alloc::HEAP.add_region(esp_alloc::HeapRegion::new(
            HEAP.as_mut_ptr() as *mut u8,
            HEAP_SIZE,
            esp_alloc::MemoryCapability::Internal.into(),
        ));
    }
}

use alloc::{vec::Vec};

#[esp_hal_embassy::main]
async fn main(_spawner: Spawner) {
    let peripherals = esp_hal::init(esp_hal::Config::default());

    init_heap();

    esp_println::logger::init_logger_from_env();

    let timg0 = esp_hal::timer::timg::TimerGroup::new(peripherals.TIMG0);
    esp_hal_embassy::init(timg0.timer1);
    let _init = esp_wifi::init(
        timg0.timer0,
        esp_hal::rng::Rng::new(peripherals.RNG),
        peripherals.RADIO_CLK,
    )
    .unwrap();

    let bluetooth = peripherals.BT;
    let mut connector = BleConnector::new(&_init, bluetooth);

    const OGF_BT_COMMAND: u8 = 0x3;
    const OCF_RESET: u16 = 0x3;
    loop {
        log::info!("Heap used: {}", esp_alloc::HEAP.used());
        log::info!("Heap free: {}", esp_alloc::HEAP.free());

        let opcode: u16 = (((OGF_BT_COMMAND & 0x3f) as u16) << 10) | (OCF_RESET & 0x3ff);
        let mut full_command: Vec<u8> = vec![];
        full_command.push(0x1);
        full_command.extend(opcode.to_le_bytes());
        full_command.push(0x0);
        if let Err(_x) = connector.write_all(&full_command) {
            log::info!("ERROR");
        }

        Timer::after(Duration::from_millis(2_000)).await
    }
}

```

## Expected behavior

It is expected that no memory gets leaked. Instead, it can be observed that the unallocated heap memory is constantly decreasing.

## Environment

- Target device: esp32 (revision v3.1)
- Crate name and version: esp-wifi 0.11.0

bjoernQ, 2024-12-16:
Thanks for creating this issue and the reproducer. 

Seems I can reproduce it with your code even on ESP32-C6 - at least I don't see anything too obvious causing it

bjoernQ, 2024-12-16:
What is happening here is you are writing a command to the controller and the controller will respond to it. But the code above never reads the response, so the data queues up.

Maybe in earlier versions the data received was overwriting older responses

Dominaezzz, 2024-12-30:
Can this be closed now? Seems like a non-issue to me.

bjoernQ, 2024-12-30:
> Can this be closed now? Seems like a non-issue to me.

Thanks for reminding me