esp-rs/esp-hal #2820: Investigate C-INTERMEDIATE for all stabilized drivers

bugadani, 2024-12-16:
cc #2784

bjoernQ, 2024-12-19:
To avoid breaking changes in future:

To resolve this I guess for the non-error case we need to carefully look at every function returning `!` or `Result<(),E>` (or where `Result::T` is a simple type)

For the error case we could consider that every error variant contains a payload, e.g. `MaxDmaTransferSizeExceeded(MaxDmaTransferSizeExceededInfo),` - initially that's just an empty struct (non-exhaustive) - if we ever consider to provide more information we could extend that struct without a breaking change

However, I'm not sure that is worth it

MabezDev, 2025-01-10:
So I believe this only applies to I2C & SPI, as they are the two drivers with the concept of a transaction.

I think to take @bjoernQ's ideas a bit further we should:

We should change the return type of spi/i2c transactions to be `Result<(), TransactionError>`, where `TransactionError` looks like:

```rust
pub struct TransactionError {
   inner: Error,
}
```

Initially we'd only expose the error inside `TransactionError`, but it would create a path for us to give some more context to the error when multiple operations may have occurred. For example adding ` sucessful_operations: usize` and a method to  access it.

The thing is, embedded-hal only allows us to have on error type for the whole driver, so this will never be available with the eh api. So here we have the option of marking the inherent `transaction` methods on the drivers as unstable, and we can punt this to post 1.0. Thoughts?

MabezDev, 2025-01-14:
In our triage we decided that it's unlikely to be useful to have intermediate results. Sending the doing the transaction  again is likely simpler, and less error prone overall.

We can expose this in a lower level transaction builder, but for now I'm closing this.

closed 2025-01-14 by MabezDev