esp-rs/esp-hal #2827: Docs fail to build, without the docs feature enabled.

Frostie314159, 2024-12-17:
## Bug description
When generating the docs for any crate, that has `esp-hal` in the dep-tree, the build fails while documenting `esp-hal/src/twai/mod.rs`, since `embedded-can` is only included when the `unstable` feature is enabled. 

## To Reproduce

1. Generate a new crate and point `esp-hal` to the main branch
2. Run `cargo doc`

## Expected behavior
Docs don't fail.

## Environment

- Any chip

bugadani, 2024-12-17:
Please use `cargo xtask build-documentation --chips <your chip> --packages esp-hal`. Also please note that the whole crate is being overhauled in a major way in this development cycle, where things may or may not work at all before we release it. Breakages like this are expected to happen, especially considering we've barely begun marking APIs unstable.

Frostie314159, 2024-12-17:
Thanks for the quick response, that resolves the question.

closed 2024-12-17 by Frostie314159