esp-rs/esp-hal #2839: Peripheral drivers should validate configuration before applying it to hardware

bugadani, 2024-12-19:
`apply_config` is fallible, which means two things:

- It can fail
- The user can ignore the failure and keep using the driver

Because of this, we should validate the whole configuration before we apply it to the hardware.

playfulFence, 2025-01-27:
When this issue is resolved, we must not forget to add the appropriate `# Errors` entries (see related FIXMEs in [this PR](https://github.com/esp-rs/esp-hal/pull/3036/files)).

MabezDev, 2025-01-27:
On further reflection, and a push from @playfulFence I think this is indeed something we want to get it sooner rather than later, for example we can set the baud to a billion bps, and [apply_config](https://github.com/esp-rs/esp-hal/blob/5139ff1a485fa997537c80b07dd701f0178e8559/esp-hal/src/uart.rs#L2187-L2198) will happily try to do it.

PR #3074 (2025-02-06): More validation and `# Errors` entries
Introduces more validation for SPI and UART, allows setting baudrate tolerance for UART config, adds more `# Errors` entries
closes #2839