esp-rs/esp-hal #2846: esp-config cannot be used with esp-hal-embassy

bugadani, 2024-12-19:
esp-hal's build script errors if users set esp-hal-embassy config options. This is, because esp-hal doesn't know about the possible options, and esp-hal is a prefix of esp-hal-embassy, therefore the crate thinks that ESP_HAL_EMBASSY_ config values belong to it, but rejects them as unknown.

MabezDev, 2024-12-19:
Yeah we should fix this ASAP, added to the milestone.

bugadani, 2024-12-19:
Any ideas how? :D I guess we can list the conflicting prefixes for esp-hal, but that solution only works as long as we don't create another esp-hal-* crate. Or even worse, someone else creates one - looking at you, community crates.

bjoernQ, 2024-12-19:
only thing I can think of (and is breaking) is to have a special separator between the crate name and the config key.

Something like `ESP_HAL_EMBASSY_CFG_config` - then `esp-hal` would look for `ESP_HAL_CFG_xxx` only

and even a crate named `esp-hal-cfg` would look for `ESP_HAL_CFG_CFG_xxx`

But it's not ideal

PR #2848 (2025-01-03): Use `_CONFIG_` to separate config prefix and key
A lazy attempt to resolve #2846
closed 2025-01-03 by MabezDev