esp-rs/esp-hal #2863: SPI DMA PSRAM QuadSPI  half_duplex_write() doesn't resolve

i404788, 2024-12-23:
## Bug description
Exact cause in not clear to me, but attempting to send a larger buffer with SPI DMA using PSRAM causes no interrupts to trigger (manually verified), and thus never resolves `is_done()`.

I can only reliably reproduce it with half_duplex_write, although I think I've also seen it with regular `write()`.

## To Reproduce

Run example: https://github.com/i404788/esp-hal/blob/spi-dma-wip/examples/src/bin/spi_loopback_dma_psram.rs

Logs:

```
 Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40 MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       68:b6:b3:31:64:e8
App/part. size:    113,824/1,048,576 bytes, 10.86%
[2024-12-23T13:22:30Z INFO ] Segment at address '0x0' has not changed, skipping write
[2024-12-23T13:22:30Z INFO ] Segment at address '0x8000' has not changed, skipping write
[00:00:04] [========================================]      44/44      0x10000                                                                                                                                       [2024-12-23T13:22:36Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 08:07:32
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.1
I (39) boot.esp32s3: Boot SPI Speed : 40MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c010020 size=03fe4h ( 16356) map
I (109) esp_image: segment 1: paddr=0001400c vaddr=3fc8a010 size=00aa8h (  2728) load
I (114) esp_image: segment 2: paddr=00014abc vaddr=40378000 size=02010h (  8208) load
I (124) esp_image: segment 3: paddr=00016ad4 vaddr=00000000 size=09544h ( 38212)
I (139) esp_image: segment 4: paddr=00020020 vaddr=42000020 size=0bc50h ( 48208) map
I (153) boot: Loaded app from partition at offset 0x10000
I (153) boot: Disabling RNG early entropy source...
INFO - Starting SPI loopback test
INFO - chip id = 1b5d0d
INFO - size is 2097152
INFO - PSRAM core_clock SpiTimingConfigCoreClock80m, flash_div = 1, psram_div = 2
INFO - TX: 0x3c020000 len 8192 (3 descripters)
INFO - First dmabuffer: Some(DmaTxBuf { descriptors: DescriptorSet { descriptors: [DmaDescriptor { flags: DmaDescriptorFlags { size: 4095, length: 4095, suc_eof: false, owner: "CPU" }, buffer: 0x3c020000, next: 0x3fc8aac4 }, DmaDescriptor { flags: DmaDescriptorFlags { size: 4095, length: 4095, suc_eof: false, owner: "CPU" }, buffer: 0x3c020fff, next: 0x3fc8aad0 }, DmaDescriptor { flags: DmaDescriptorFlags { size: 2, length: 2, suc_eof: false, owner: "CPU" }, buffer: 0x3c021ffe, next: 0x0 }] }, buffer: [0, 0, 32, 0, 0, 0, 0, 0, 8, 9, 10, 11, 12, 13, 14, 14, 0, 17, 18, 18, 148, 21, 18, 19, 24, 9, 26, 59, 28, 31, 30, 31, 34, 33, 34, 163, 40, 37, 38, 39, 40, 41, 42, 171, 44, 45, 174, 43, 48, 57, 50, 50, 36, 53, 38, 167, 56, 57, 58, 58, 60, 61, 58, 63, 64, 65, 66, 3, 68, 69, 66, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 80, 82, 83, 68, 85, 210, 87, 88, 89, 90, 91, 76, 93, 78, 95, 96, 97, 98, 98, 100, 97, 102, 238, 104, 105, 106, 235, 110, 108, 110, 111, 120, 97, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 60, 125, 126, 127, 128, 128, 130, 131, 132, 133, 134, 135, 136, 137, 138, 138, 142, 141, 142, 174, 128, 145, 146, 147, 148, 149, 134, 146, 136, 153, 154, 139, 156, 157, 158, 191, 160, 161, 162, 163, 164, 165, 166, 162, 170, 170, 170, 170, 168, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 168, 171, 170, 187, 190, 173, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 206, 207, 238, 239, 192, 209, 210, 211, 196, 213, 214, 191, 216, 217, 218, 219, 204, 221, 222, 223, 224, 233, 234, 227, 228, 229, 230, 231, 232, 233, 234, 234, 238, 237, 238, 239, 240, 249, 242, 243, 240, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 8, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 10, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 7, 8, 24, 26, 27, 24, 29, 26, 31, 32, 33, 34, 35, 32, 37, 38, 39, 40, 41, 170, 42, 44, 41, 46, 47, 48, 49, 186, 51, 52, 53, 54, 55, 40, 56, 58, 58, 44, 45, 58, 63, 64, 65, 66, 67, 68, 69, 70, 70, 72, 73, 74, 75, 76, 13, 78, 15, 80, 65, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 76, 93, 94, 95, 96, 41, 98, 99, 100, 101, 102, 103, 40, 105, 106, 107, 108, 106, 110, 111, 112, 113, 114, 115, 116, 119, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 129, 134, 135, 168, 137, 138, 171, 140, 141, 142, 143, 128, 145, 146, 147, 132, 149, 150, 151, 136, 153, 154, 155, 140, 153, 154, 190, 160, 161, 162, 163, 160, 165, 162, 167, 168, 169, 170, 170, 172, 173, 174, 174, 176, 177, 178, 179, 180, 165, 182, 178, 184, 169, 186, 187, 168, 173, 190, 191, 192, 193, 194, 195, 196, 197, 194, 199, 200, 201, 202, 203, 200, 205, 206, 207, 208, 193, 210, 211, 212, 213, 246, 223, 200, 201, 218, 219, 220, 221, 254, 223, 232, 225, 162, 227, 238, 229, 174, 230, 234, 235, 234, 234, 236, 237, 238, 239, 240, 163, 242, 243, 160, 245, 246, 247, 232, 233, 250, 250, 236, 253, 254, 255, 0, 1, 2, 3, 4, 5, 6, 135, 8, 9, 10, 11, 12, 13, 14, 14, 0, 1, 18, 18, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 158, 63, 32, 33, 162, 35, 40, 37, 38, 38, 40, 41, 42, 43, 44, 45, 174, 47, 48, 49, 34, 50, 32, 53, 54, 39, 56, 57, 58, 59, 60, 61, 62, 191, 64, 1, 66, 67, 68, 69, 66, 71, 72, 9, 74, 75, 76, 77, 78, 78, 80, 81, 82, 83, 84, 85, 214, 87, 88, 89, 90, 91, 76, 95, 94, 95, 96, 97, 98, 99, 100, 33, 102, 230, 104, 105, 106, 107, 108, 108, 110, 111, 120, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 60, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 138, 140, 141, 142, 174, 144, 145, 146, 147, 148, 149, 134, 147, 136, 153, 154, 139, 156, 157, 158, 187, 160, 161, 162, 163, 172, 165, 166, 162, 168, 171, 170, 170, 168, 169, 174, 175, 176, 177, 178, 179, 180, 181, 182, 182, 184, 169, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 206, 207, 206, 207, 208, 209, 210, 211, 196, 213, 246, 223, 216, 217, 250, 219, 204, 221, 222, 223, 224, 169, 234, 227, 228, 229, 230, 231, 232, 233, 234, 234, 236, 237, 238, 239, 240, 241, 242, 243, 244, 245, 246, 247, 248, 249, 250, 251, 252, 253, 254, 255, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 10, 14, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 6, 8, 25, 26, 27, 24, 29, 26, 31, 32, 33, 34, 35, 36, 37, 38, 35, 40, 41, 170, 42, 44, 41, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 58, 58, 58, 60, 44, 186, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 64, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 33, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 111, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 62, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 136, 153, 154, 155, 156, 157, 158, 158, 160, 161, 162, 163, 160, 165, 162, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 182, 184, 169, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 134, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 223, 216, 201, 218, 219, 220, 221, 222, 223, 232, 225, 226, 227, 228, 229, 230, 231, 232, 235, 234, 234, 236, 237, 238, 239, 240, 225, 242, 243, 180, 245, 246, 247, 248, 249, 250, 250, 236, 253, 254, 254, 0, 1, 2, 3, 4, 5, 6, 135, 8, 9, 10, 11, 12, 141, 14, 142, 0, 1, 18, 146, 20, 21, 22, 55, 24, 25, 26, 27, 28, 29, 30, 63, 34, 33, 34, 35, 40, 37, 38, 39, 40, 41, 42, 43, 44, 45, 174, 43, 48, 49, 34, 50, 36, 53, 38, 39, 40, 57, 186, 59, 60, 61, 62, 63, 64, 1, 66, 67, 68, 69, 66, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 68, 85, 210, 87, 88, 89, 90, 91, 76, 93, 94, 95, 96, 97, 98, 99, 100, 33, 102, 238, 104, 105, 106, 10 [... cut ...] 42, 243, 164, 245, 246, 247, 234, 249, 250, 250, 232, 253, 254, 255], burst: BurstConfig { external_memory: Size64, internal_memory: Disabled } })
0x3fc8aac4 - spi_loopback_dma_psram::__xtensa_lx_rt_main::DESCRIPTORS
    at ??:??
0x3fc8aad0 - spi_loopback_dma_psram::__xtensa_lx_rt_main::DESCRIPTORS
    at ??:??
```

## Expected behavior

Expected it to either error, or complete successfully. See the `Done` log repeatedly.

## Environment

- Target device: ESP32-S3]
- Crate name and version: main

i404788, 2024-12-23:
Update: found that the bus speed is relevant: <20Mhz still works 25Mhz fails

Dominaezzz, 2024-12-23:
Thanks for the tidbit about bus speed. It's possible that PSRAM can't keep up with the SPI.

Would you mind printing out the SPI interrupt registers (with the PAC) a few seconds after the transfer starts?

i404788, 2024-12-23:
@Dominaezzz attempted to add your change but it seems adding a delay of >600us (bisected to +-50us, in this range it occasionally works) causes it to succeed again. So it seems like there might be a race condition with an interrupt or something.

The result I'm getting is always `0` but I may be doing it wrong:
```rust
      log::info!("SPI2: {}", unsafe {
          *(SPI2::ptr().add(0x0040) as *const u32).as_ref().unwrap()
      });
```

Dominaezzz, 2024-12-23:
I'm looking for INT_RAW rather than INT_ST, which is `0x003C`.

Did you add the delay to the example or to the SPI driver? I was thinking of [here](https://github.com/i404788/esp-hal/blob/spi-dma-wip/examples/src/bin/spi_loopback_dma_psram.rs#L135).

Since this example is blocking and not async, a race condition with an interrupt is very unlikely.

i404788, 2024-12-23:
> Did you add the delay to the example or to the SPI driver? I was thinking of [here](https://github.com/i404788/esp-hal/blob/spi-dma-wip/examples/src/bin/spi_loopback_dma_psram.rs#L135).

Yes exactly there, I've pushed my latest version now.

Dominaezzz, 2024-12-23:
Can you add a larger delay? I still want to see what the interrupts look like if it succeeds

i404788, 2024-12-23:
Hmm switched to `read_volatile` but still no change either way:
```rust
        log::info!("pre-SPI2: {}", unsafe {
            core::ptr::read_volatile(SPI2::ptr().add(0x003C) as *const u32)
        });

        delay.delay_millis(2000);

        log::info!("post-SPI2: {}", unsafe {
            core::ptr::read_volatile(SPI2::ptr().add(0x003C) as *const u32)
        });
        (spi, dma_buf) = transfer.wait();
```

```
INFO - pre-SPI2: 0
INFO - post-SPI2: 0
```

However it does not resolve again anymore. Given that; maybe it has to do with memory layout since that will only be randomized with some significant change? This seems to be confirmed by placing the delay between the prints also fails with 593us, even though placing it before the prints will succeed once then fail once. Likewise placing 750us between prints fails, placing it before the prints get repeated successes.

I'm not sure which parts require a specific alignment and how I can check if the are aligned.

Dominaezzz, 2024-12-23:
The alignment requirements should be checked by `DmaTxBuf`, which Daniel improved the checks for recently.

Can you print out the addresses of `tx_descriptors` and `tx_buffer`? We need to make sure the latter is 64 byte aligned.

Dominaezzz, 2024-12-23:
Oh looks like it's already done here.
```
    info!(
        "TX: {:p} len {} ({} descripters)",
        tx_buffer.as_ptr(),
        tx_buffer.len(),
        tx_descriptors.len()
    );
```

i404788, 2024-12-23:
Hmm yeah the descriptors/buffer addresses don't seem to different:

Working:
```
INFO - TX: 0x3c020000 len 8192 (3 descripters)
INFO - Alignment: 64, Chunk size: 4032
INFO - First dmabuffer: Some(DmaTxBuf { descriptors: DescriptorSet { descriptors: [DmaDescriptor { flags: DmaDescriptorFlags { size: 4095, length: 4095, suc_eof: false, owner: "CPU" }, buffer: 0x3c020000, next: 0x3fc8aacc }, DmaDescriptor { flags: DmaDescriptorFlags { size: 4095, length: 4095, suc_eof: false, owner: "CPU" }, buffer: 0x3c020fff, next: 0x3fc8aad8 }, DmaDescriptor { flags: DmaDescriptorFlags { size: 2, length: 2, suc_eof: false, owner: "CPU" }, buffer: 0x3c021ffe, next: 0x0 }] }
```

Failing:
```
INFO - TX: 0x3c020000 len 8192 (3 descripters)
INFO - Alignment: 64, Chunk size: 4032
INFO - First dmabuffer: Some(DmaTxBuf { descriptors: DescriptorSet { descriptors: [DmaDescriptor { flags: DmaDescriptorFlags { size: 4095, length: 4095, suc_eof: false, owner: "CPU" }, buffer: 0x3c020000, next: 0x3fc8aacc }, DmaDescriptor { flags: DmaDescriptorFlags { size: 4095, length: 4095, suc_eof: false, owner: "CPU" }, buffer: 0x3c020fff, next: 0x3fc8aad8 }, DmaDescriptor { flags: DmaDescriptorFlags { size: 2, length: 2, suc_eof: false, owner: "CPU" }, buffer: 0x3c021ffe, next: 0x0 }] }, 
```

i404788, 2024-12-23:
After much testing it's seems to be an very perplexing issue. Ghidra shows effectively no difference between a working example and a broken example (apart from the delay moving):
[image]

Also of note it seems debug builds are always broken, esp-hal@main requires 1.83 so I haven't been able to test the previous compiler versions.

Further looking through Ghidra, some of the data sections do have a layout shift of 3:
[image]

So the only conclusion I can make is that,  the layout of some unknown data (non-descriptor non-buffer?) is important.

Dominaezzz, 2024-12-23:
I've run out of ideas for what could be going on but as a final hail mary, try adding a delay to the update function [here](https://github.com/esp-rs/esp-hal/blob/36095e447e6b99115c34e591f3c29e292638b9e0/esp-hal/src/spi/master.rs#L3289). You can copy that into the gdma section and make it bigger, like 10 instead of just 1.

bjoernQ, 2024-12-24:
I tried the linked code and can reproduce the issue.

When I change `DataMode` to `Dual` it works fine - even at higher frequencies. On a device with octal-PSRAM it also works fine (with `Quad`)

Placing most of the involved code into RAM lets me go up to 32MHz
(maybe not all is needed but I used `ESP_HAL_PLACE_SWITCH_TABLES_IN_RAM`, `ESP_HAL_PLACE_ANON_IN_RAM`, `ESP_HAL_PLACE_SPI_DRIVER_IN_RAM`, put delay_micros, wait and main into RAM, removing the logs before `wait` )

Seems to be a bandwidth issue (i.e. placing code in RAM "eliminates" concurrently accessing flash)

i404788, 2024-12-24:
I'll try both suggestions and report back. @bjoernQ assuming it's the same issue I encountered; any chance it can be detected; I'd rather it panic or Err than hang.

i404788, 2024-12-24:
Hmm I put everything relevant in ram removed logs before wait, and added:
```
ESP_HAL_PLACE_SPI_DRIVER_IN_RAM = "true"
ESP_HAL_PLACE_ANON_IN_RAM = "true"
ESP_HAL_PLACE_SWITCH_TABLES_IN_RAM = "true"
```

However the env variables actually causes it to fail (commenting succeeds again). If it is flash contention, it seems like it would be at the `half_duplex_write()`-side rather than the `wait()` side, since placing a log after the delay works, but before doesn't.

I'll see if I can get a minimal break condition (e.g. call to an empty function or similar) to maybe get a better look in ghidra.

@Dominaezzz I tried your hail-mary, unfortunately doesn't seem to change anything.

i404788, 2024-12-24:
I've reduced the break condition to a few `vec::push`es: https://github.com/i404788/esp-hal/blob/spi-dma-wip/examples/src/bin/spi_loopback_dma_psram2.rs (note new file).  Even with black_box, ghidra shows large code shifts when changing the hard-coded bool which makes it hard to compare, I tried using bindiff but it seems like the ghidra plugin is broken on xtensa.

I've also tried manually allocating and generating some PSRAM traffic (see commented fail code) but that didn't seem to matter much. 

In the fail=false condition, it just works.

In the fail=true condition **without tuning** I'm getting the original failure mode (infinite wait).
In the fail=true condition **with the ram envs** i'm getting:
```
====================== PANIC ======================
panicked at /home/null/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/panicking.rs:221:5:
unsafe precondition(s) violated: slice::from_raw_parts requires the pointer to be aligned and non-null, and the total size of the slice not to exceed `isize::MAX`

Backtrace:

0x4200b878
0x4200b878 - core::slice::raw::from_raw_parts::precondition_check
    at /home/null/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ub_checks.rs:70
0x4200b586
0x4200b586 - <esp_println::Printer as core::fmt::Write>::write_str
    at /home/null/Projects/esp-hal-spi/esp-println/src/lib.rs:82
0x42005237
0x42005237 - core::fmt::Formatter::pad
    at /home/null/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/mod.rs:1438
0x4200bb88
0x4200bb88 - <log::Level as core::fmt::Display>::fmt
    at /home/null/.cargo/registry/src/index.crates.io-6f17d22bba15001f/log-0.4.22/src/lib.rs:534
0x42005950
0x42005950 - core::fmt::rt::Argument::fmt
    at /home/null/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/rt.rs:177
0x4200b54c
0x4200b54c - <&mut W as core::fmt::Write::write_fmt::SpecWriteFmt>::spec_write_fmt
    at /home/null/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/mod.rs:226
0x42003372
0x42003372 - log::__private_api::log_impl
    at /home/null/.cargo/registry/src/index.crates.io-6f17d22bba15001f/log-0.4.22/src/__private_api.rs:61
0x4200c037
0x4200c037 - Reset
    at /home/null/Projects/esp-hal-spi/xtensa-lx-rt/src/lib.rs:82
0x40379aa6
0x40379aa6 - ESP32Reset
    at /home/null/Projects/esp-hal-spi/esp-hal/src/soc/esp32s3/mod.rs:164
```

bjoernQ, 2024-12-24:
The panic looks sus - we have seen some miscompiles / misoptimizations lately.

However I tried your changed code on 36095e447e6b99115c34e591f3c29e292638b9e0 plus my changes and some more changes and it seems to work

<details>

<summary>Here are my changes (on top of the latest commit on main)</summary>

```diff
diff --git a/esp-hal/src/delay.rs b/esp-hal/src/delay.rs
index a361aabc..5fb26144 100644
--- a/esp-hal/src/delay.rs
+++ b/esp-hal/src/delay.rs
@@ -68,6 +68,7 @@ impl Delay {
     }
 
     /// Delay for the specified number of microseconds
+    #[procmacros::ram]
     pub fn delay_micros(&self, us: u32) {
         let delay = MicrosDurationU64::micros(us as u64);
         self.delay(delay);
diff --git a/esp-hal/src/spi/master.rs b/esp-hal/src/spi/master.rs
index 48116c03..6e3d6d01 100644
--- a/esp-hal/src/spi/master.rs
+++ b/esp-hal/src/spi/master.rs
@@ -1316,6 +1316,7 @@ mod dma {
         /// This method blocks until the transfer is finished and returns the
         /// `SpiDma` instance and the associated buffer.
         #[instability::unstable]
+        #[procmacros::ram]
         pub fn wait(mut self) -> (SpiDma<'d, Dm, T>, Buf) {
             self.spi_dma.wait_for_idle();
             let retval = unsafe {
@@ -1581,6 +1582,7 @@ mod dma {
         #[allow(clippy::type_complexity)]
         #[cfg_attr(place_spi_driver_in_ram, ram)]
         #[instability::unstable]
+        #[ram]
         pub fn half_duplex_write<TX: DmaTxBuffer>(
             mut self,
             data_mode: DataMode,
diff --git a/examples/.cargo/config.toml b/examples/.cargo/config.toml
index 600f74b2..d25c3d15 100644
--- a/examples/.cargo/config.toml
+++ b/examples/.cargo/config.toml
@@ -35,5 +35,9 @@ GATEWAY_IP = "1.1.1.1"
 HOST_IP = "1.1.1.1"
 ESP_WIFI_CSI_ENABLE = "true"
 
+ESP_HAL_PLACE_SPI_DRIVER_IN_RAM = "true"
+ESP_HAL_PLACE_ANON_IN_RAM = "true"
+ESP_HAL_PLACE_SWITCH_TABLES_IN_RAM = "true"
+
 [unstable]
 build-std = ["alloc", "core"]
diff --git a/examples/src/bin/spi_loopback_dma_psram.rs b/examples/src/bin/spi_loopback_dma_psram.rs
index 41b785b4..61937507 100644
--- a/examples/src/bin/spi_loopback_dma_psram.rs
+++ b/examples/src/bin/spi_loopback_dma_psram.rs
@@ -16,13 +16,16 @@
 //! If your module is quad PSRAM then you need to change the `psram` feature in the
 //! in the features line below to `quad-psram`.
 
-//% FEATURES: esp-hal/log esp-hal/octal-psram esp-hal/unstable
+//% FEATURES: esp-hal/log esp-hal/quad-psram esp-hal/unstable
 //% CHIPS: esp32s3
 
 #![no_std]
 #![no_main]
+#![feature(vec_push_within_capacity)]
[... 170 lines cut ...]
-            if dma_rx_buf.as_slice()[i] != *v {
-                error!(
-                    "Mismatch at index {}: expected {}, got {}",
-                    i,
-                    *v,
-                    dma_rx_buf.as_slice()[i]
-                );
-                break;
-            }
-        }
-        info!(
-            "{:0x?} .. {:0x?}",
-            &dma_rx_buf.as_slice()[..10],
-            &dma_rx_buf.as_slice().last_chunk::<10>().unwrap()
-        );
-        dma_tx_buf.as_mut_slice().reverse();
         delay.delay_millis(1000);
     }
-}
+}
\ No newline at end of file
```
</details>

(I copied your new example as the original one)

i404788, 2024-12-24:
@bjoernQ Can confirm your example works (similar to fail=false in my case), however I'm a bit concerned on the fragility. Adding the log back will halt it again, and so the thing I'm trying to find out is if `half_duplex_write().wait()` always works or if it just happens to work in this exact configuration. And those experiments do point toward fairly consistent miscompiles/layout mismatches, but I'll see if I can unexpectedly break it without inserting anything in between.

i404788, 2024-12-24:
Yeah it doesn't work in the project I originally found the issue (although I can't set `ESP_HAL_PLACE_ANON_IN_RAM` because it breaks embassy) even without anything in-between.

i404788, 2025-01-20:
Update: it seems like this happens always if the CpuClock is not the default. E.g. CpuClock@160MHz Spi@500kHz half_duplex_write in Single DataMode *will not resolve* (even though I can see it send all the data). However going back to the default clock of 80Mhz (explicit or not) it works till Spi@23Mhz.

Investigating a bit it seems like https://github.com/esp-rs/esp-hal/blob/d1ee2ba733e32fe15ea084765b48a9844e28e749/esp-hal/src/clock/clocks_ll/esp32s3.rs#L8 is only called if the clock is non-default

This is also reproducable in the example: https://github.com/i404788/esp-hal/blob/spi-dma-test/examples/src/bin/spi_loopback_dma_psram.rs; so it could be bus contention (but it would be primarily caused by the CPU since SPI & DMA are using the APB clock directly) or it could be some misconfiguration although I'm not sure what it would be. `SYSTEM_CPU_WAIT_MODE_FORCE_ON` seems like it would match somewhat but it should be set by default

i404788, 2025-01-20:
Seemingly fixed the 240Mhz issue by adding a 1us delay anywhere in here: https://github.com/esp-rs/esp-hal/blob/d1ee2ba733e32fe15ea084765b48a9844e28e749/esp-hal/src/spi/master.rs#L2431-L2436

Requires more validation required but supposedly `prepare_transfer`/`start_transfer` needs some setup time before `start_operation` can run.

bugadani, 2025-01-20:
Neat, this is the sort of weirdness I intend to debug for https://github.com/esp-rs/esp-hal/issues/2909 as I believe they are related. I don't know how, yet, but we'll need to understand because random delays are a bandaid, not a solution.

Dominaezzz, 2025-01-20:
This is still from PSRAM right? It's likely it needs a larger delay compared to internal ram.

Dominaezzz, 2025-01-20:
> I don't know enough about the hw internals but does the GDMA engine need to be checked if it's ready?

Sort of. The DMA needs a bit of time to populate the peripheral's FIFO.
Some peripherals on some chips do have a way to check if it's ready to go but I don't think SPI on the S3 can do it iirc.

Unfortunately this isn't a very well documented quirk.

MabezDev, 2025-01-21:
@i404788 Once https://github.com/esp-rs/esp-hal/pull/3003 is merged, would you mind trying again from main?

i404788, 2025-01-21:
@MabezDev Yes, would love to test it out, I can also test it from the PR directly if you want to know before merging.

i404788, 2025-01-22:
Maybe slightly off-topic but it seems like after `half_duplex` it doesn't seem possible to go back to full-duplex mode.
I've tried resetting the relevant registers before the dma transfer but it still always receives zeros: https://github.com/i404788/esp-hal/commit/0796e011d19a6566398fa0b4a1e32dbc4c932a88

Dominaezzz, 2025-02-09:
> after `half_duplex` it doesn't seem possible to go back to full-duplex mode.

I'm curious whether re-creating the driver makes it work. I wouldn't be surprised if there was some register somewhere that's not being overwritten correctly when doing full duplex.

Mind opening a new issue for that? Seems like this one has been resolved.

i404788, 2025-02-10:
Yeah I'll close this one and make a new one for the switch-over. I don't expect re-creating to change much since went through all of the function bodies to do the registers but maybe I missed an important one somewhere.

closed 2025-02-10 by i404788