esp-rs/esp-hal #2865: Move `malloc()`, `free()`, etc. from `esp-wifi` to `esp-alloc`

AnthonyGrondin, 2024-12-24:
## Motivations

`esp-wifi`, `esp-alloc` and `esp-mbedtls` are tightly coupled by their memory allocator implementation. `esp-wifi` needs an implementation of `malloc()` and `free()`, and since #2099, it uses Rust's Global allocator. `esp-alloc` is a provider of a global allocator for ESP SoCs but users are free to provide their own allocator if they want.

`esp-mbedtls` and `esp-mbedtls-sys` depend on `esp-wifi`, strictly for providing `malloc()` and `free()`. 

## Solution

These should be moved to `esp-alloc`, to only depend on an allocation crate, if one only needs `malloc()`, `free()`, etc. They should be defined with conditional compilation and / or weak linkage to avoid conflict issues if another crate provides these symbols.

## Alternatives

Provide a `c-lib` / `c-compat` / `esp-c-compat` crate to define and implement C symbols, with conditional compilation. It would not only provide memory allocation symbols, but any symbols such as `vsnprintf()`.

## Additional context

- https://github.com/esp-rs/esp-hal/issues/2445
- https://github.com/esp-rs/esp-hal/pull/2183
- https://github.com/esp-rs/esp-mbedtls/blob/1431835a9a3464820e97203347ec33e313f8e116/esp-mbedtls-sys/Cargo.toml#L18-L22

Dominaezzz, 2024-12-24:
Some extra motive. `esp-wifi` won't be available on the P4 (No WiFi or Bluetooth) but there's no reason `esp-mbedtls` shouldn't work there. The hard dependency on `esp-wifi` gets in the way.

AnthonyGrondin, 2024-12-24:
Ideally `esp-mbedtls` shouldn't depend on any ESP crates, because in the end, it's just a transport layer, over `embedded-io(-async)` Read and Write. We will provide hardware acceleration and such using crypto providers, the same way `rustls` and `embedded-tls` does. With a few refactors, it will be completely platform agnostic, and can work on any transport medium.

ivmarkov, 2025-01-16:
In my oppinion, `esp-mbedtls` depending on `esp-alloc` is not ideal either, as this way, the crate depends on another (esp specific or at least heap _implementation_-specific) crate: `esp-alloc`.

`esp-mbedtls` should only depend/assume there is a global allocator and nothing else.

If `esp-mbedtls` needs to implement `calloc`/`free` on top of the global allocator crate, it needs to copy-paste the `calloc`/`free` impls from `esp-wifi` which do exactly that.

This will not pollute the global C namespace with `calloc` and `free`, because mbedtls has its own `calloc` and `free` hooks that we should use anyway.

AnthonyGrondin, 2025-01-17:
> In my oppinion, `esp-mbedtls` depending on `esp-alloc` is not ideal either, as this way, the crate depends on another (esp specific or at least heap _implementation_-specific) crate: `esp-alloc`.
> 
> `esp-mbedtls` should only depend/assume there is a global allocator and nothing else.
> 
> If `esp-mbedtls` needs to implement `calloc`/`free` on top of the global Rust allocator API, it needs to copy-paste the `calloc`/`free` impls from `esp-wifi` which do exactly that.
> 
> This will not pollute the global C namespace with `calloc` and `free`, because mbedtls has its own `calloc` and `free` hooks that we should use anyway.

Totally agree with this approach. We should provide our own implementation in `esp-mbedtls` based on the global allocator to make it platform agnostic. I've been looking at it lately, into what are the best practices to implement `malloc()` / `free()` on top of Rust allocator for FFI.

ivmarkov, 2025-01-17:
> In my oppinion, `esp-mbedtls` depending on `esp-alloc` is not ideal either, as this way, the crate depends on another (esp specific or at least heap _implementation_-specific) crate: `esp-alloc`. [...]

Great that we are aligned.

> Totally agree with this approach. We should provide our own implementation in `esp-mbedtls` based on the global allocator to make it platform agnostic. I've been looking at it lately, into what are the best practices to implement `malloc()` / `free()` on top of Rust allocator for FFI.

I think there are not many as this is the "reverse" case (Cheap-API-on-top-of-RustHeap-API) which is very rare, compared to the standard RustHeap-API-on-top-of-Cheap-API. Where the latter is by the way done with `posix_memalign` and not with regular `malloc` because `malloc` usually does not align to more than the "natural" alignment for the architecture, while Rust sometimes needs more.

I think other than storing the size of the allocated block in front of the return buffer and then always aligning to 4 or 8 bytes (what @bjoernQ does anyway) there are not very many other options? At least I couldn't find an alternative.

bjoernQ, 2025-06-10:
Besides the potential alignment issue, is there anything left to do here?

I'm considering using weak-linkage for `malloc`, `calloc`, `realloc` and `free` in esp-wifi but not sure it's worth it?

closed 2025-08-05 by bjoernQ