esp-rs/esp-hal #2883: esp-wifi: Examples broken

bjoernQ, 2025-01-03:
On current main

- `wifi_embassy_access_point`: DHCP server is not working anymore
- Enterprise wifi connect seems to be broken (no example in the repo - used a local one - crashing with 'Load access fault')

bjoernQ, 2025-01-03:
Enterprise wifi broke with commit ca9ee23b5e8ea059078af49a8365f328589aa3ff

tommasoclini, 2025-01-10:
Is this something that has to do with my example code?

bjoernQ, 2025-01-10:
> Is this something that has to do with my example code?

I don't think so. Needs some more investigation

bjoernQ, 2025-01-10:
Enterprise Wifi is fixed

bjoernQ, 2025-01-14:
Thanks. Just tried on latest commit (571760884bf90240fdace3afdfb7f527194681aa) and `wifi_embassy_ble` example is working fine for me on ESP32-C6 (on both, "chip revision: v0.1" and "chip revision: v0.0"), ran the example multiple times compiled with different toolchains

jamessizeland, 2025-01-14:
Ok `espflash erase-flash` has revived the chips, so I'll try to reproduce the fault again.

eloycoto, 2025-01-14:
> Thanks. Just tried on latest commit ([5717608](https://github.com/esp-rs/esp-hal/commit/571760884bf90240fdace3afdfb7f527194681aa)) and `wifi_embassy_ble` example is working fine for me on ESP32-C6 (on both, "chip revision: v0.1" and "chip revision: v0.0"), ran the example multiple times compiled with different toolchains

Fails for me on the second flash, first one correct, second one fails.

jamessizeland, 2025-01-14:
> Thanks. Just tried on latest commit ([5717608](https://github.com/esp-rs/esp-hal/commit/571760884bf90240fdace3afdfb7f527194681aa)) and `wifi_embassy_ble` example is working fine for me on ESP32-C6 [...]

Ok, I'm reproducibly getting the stall state by:

```bash
cargo espflash erase-flash
# unplug to depower
cargo xtask run-example esp-hal esp32c6 wifi_embassy_ble
# works as expected
# ctrl+r to reset the board or ctrl+c and reflash
# stalls at https://github.com/esp-rs/esp-hal/blob/571760884bf90240fdace3afdfb7f527194681aa/esp-wifi/src/timer/timer_esp32c6.rs#L29-L32
```

bjoernQ, 2025-01-15:
Aha ...... it always works fine for me when flashing/running via UART-bridge, but I see the same thing happening when flashing/running via Serial-JTAG - that's a hint.

Even funnier - even while connected to Serial-JTAG: stopping espflash and resetting the board makes the device show up in BLE Explorer etc. (and I can connect)

bjoernQ, 2025-01-15:
But e.g. `wifi_ble` example should show the same behavior, then. (and does for me - but wifi examples don't show this behavior) 🤔

eloycoto, 2025-01-16:
@bjoernQ  Do you have any idea where the problem can be? I'm a bit lost on the possible causes, so any hint, I'm happy to check if that it's the problem.

bjoernQ, 2025-01-16:
What dev-board are you using? I'm using official DevKitC - and is it using a UART-bridge or JTAG-Serial?

eloycoto, 2025-01-16:
waveshare devkit and using JTAG

jamessizeland, 2025-01-16:
And I'm seeing the same behaviour on the Unexpected Maker Tinyc6 with the default interface, which I assume is JTAG.

bjoernQ, 2025-01-16:
Ok - I can definitely see weird behavior when serial-JTAG is used. I can't explain why it happens, but I know how to work around that.
I'm on it

bjoernQ, 2025-01-16:
#2981 fixes what I observed and (hopefully) will fix the problems you are facing (if not you encountered more than one problem - it's definitely part of the problems you see)

If possible, would be great if you can give that branch a try

jamessizeland, 2025-01-16:
Confirmed working reliably on both of my c6 boards! @bjoernQ you are a genius.

eloycoto, 2025-01-16:
I can confirm that it's working 🎉  Thanks!

PR #2981 (2025-01-17): esp-wifi: Late enable BT interrupts
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] I [... cut]
closed 2025-01-17 by MabezDev