esp-rs/esp-hal #2884: Implement `wait_for_done` for `DpiTransfer`

EliteTK, 2025-01-03:
## Motivations

`I8080Transfer` has this facility already, so this would introduce parity.

In general it can be useful when animating content on a display to avoid re-drawing more often than is necessary to ensure a smooth animation.

## Solution

An async `wait_for_done` function on the `DpiTransfer` in the same style as the one on `I8080Transfer`.

## Alternatives

I think it should be possible to achieve the same effect by simply implementing an appropriate `DmaTxBuffer` which tracks how much data has been transferred and synchronizes rendering to which frame is currently being requested combined with using `Dpi::send` with `next_frame_en` as `true`. But I think both styles would be appropriate to support.

## Additional context

I've discussed this with @Dominaezzz on matrix (in the Embassy channel) and he recommended I submit this feature request.

I have looked into this and may consider contributing this but please don't hold your breath as I think the `DmaTxBuffer` route would personally be preferable at the moment.

Dominaezzz, 2025-01-19:
We can't provide a `wait_for_done` but can provide a `wait_for_vsync`.

Dominaezzz, 2026-01-29:
I have no ideas about the wait and is_done stuff. Sounds like a weird compiler thing but idk

> 4\. and shifts right from time to time - I guess that's PSRAM issues.

Correct

> 4\. On top of that I also see the display changes appear in somewhat strange way (so new pixels appear slowly and not in order they arrived to the data)

Can you share a video of this?

Dominaezzz, 2026-01-29:
Oh actually, I can think of one reason. You might have the wrong clock phase for your screen perhaps? Try changing this config https://docs.espressif.com/projects/rust/esp-hal/1.0.0-rc.1/esp32s3/esp_hal/lcd_cam/lcd/dpi/struct.Config.html#method.clock_mode

yanshay, 2026-02-02:
> This is PSRAM bandwidth issue at it purest. The DMA isn't able to deliver the data on time and the LCD_CAM does not wait and just sends garbage until the DMA starts delivering data again. This temporary garbage is what causes the offset because the dpi doesn't fix it's counters. Looks like you're getting 8 pixels worth of garbage every so often.
> 
> There's no "fix" and I don't think there's a reliable way to detect this.
> 
> You can choose to restart the DMA transfer every few seconds to realign the display. Esp-idf offers something like this.
> 
> Or you can start looking into bounce buffers 😄

And it is so consistent after exactly 23 seconds every time (tried tens of times)? and then minutes after, many minutes.
Isn't it more reasonable to be related to some clock offset that drifts with time and then cause the shift?
The DMA doesn't restart from beginning of buffer every time there is vsync? If I had the vsync I guess I could trigger the transaction every vsync, right?

How can I do this bounce buffer when I have no signals about where the DMA is? is there access to the DMA interrups or something (as the dpi drives the DMA if I get it right, not my code).

Dominaezzz, 2026-02-02:
> I believe it has to do with embassy in some way.

I won't be surprised if it's due to the scheduler code executing from flash, which is stealing bandwidth from the DMA.

yanshay, 2026-02-03:
@EliteTK  I shared the repo at https://github.com/yanshay/esp32s3-rgb-async
It has several branches V1, V2 etc. with some comments. V3 is the latest with vsync interrupts, earlier versions were with cyclic DMA descriptors. I hope the branches arrived well. I'm using jujutsu so don't know if did everything correctly to create and push the branches.
Pretty messy code but hope you'll be able to deal with it. Let me know if you have questions.

If you have an xip (so running code from PSRAM) that I can try with esp-hal 1.0, I'd be happy to try it out and see if it helps.

Also, if you can point me to example of complex ESP32 C code with large display that use PSRAM in parallel to driving RGB display, I'd appreciate it. I couldn't find any. I only found applications showing some sensor data or control panels to other systems. Nothing that both did UI and meaningful processing at the same time with large amounts of data, TLS, etc. that would require PSRAM both for UI and for the application.

Dominaezzz, 2026-02-03:
> I don't fully understand - how does moving code to PSRAM solves this? It still needs code to run over the same SPI , so why does it solve things? Because flash is so much slower?

This was explained by Igor in a discussion but that's lost now. It had to do with arbitration. There's hints in the esp-idf docs.

yanshay, 2026-02-04:
This is what I did:
```
    lcd_cam
        .lc_dma_int_ena()
        .modify(|_, w| w.lcd_vsync_int_ena().set_bit());

    lcd_cam
        .lc_dma_int_ena()
        .modify(|_, w| w.lcd_trans_done_int_ena().set_bit());

```
Both interrupts took place, just the lcd_trans_done happened only when used non cyclic descriptors.
Yes, I used to set the eof. it had no effect. I even think I received the lcd_trans_done interrupt on send when it wasn't cyclic without it though not sure about that. 
```
unsafe { TX_DESCRIPTORS[NUM_DMA_DESC-1].set_suc_eof(true); };
```
I also tried several other descriptors and got nothing from them never.

So I don't understand how it will work - how do I know when to trigger the DMA from PSRAM to RAM? Unless there is a way to get the interrupts. 

You also said something above about a flag in the descriptor that change? what flag? Maybe I need to set it every time it fires? (though I think it never fired on cyclic descriptors).

yanshay, 2026-02-15:
Great to see I'm not the only one trying to get RGB working. What display size are trying to use?

As for your question - your included snippets are not full so hard to tell, but it is very likely Drop. I had this issue as well and eventually found something dropped without me noticing. 

Is your attempt at bounce buffers working right now?

Limeth, 2026-02-15:
> What display size are you trying to use?

I am using [this 240x960](https://www.adafruit.com/product/5799) display, but it has a front porch of 120 pixels so it actually requires a frame of 360x960.

> As for your question - your included snippets are not full so hard to tell, but it is very likely Drop. I had this issue as well and eventually found something dropped without me noticing.

Could be, but I really don't see why it would be getting dropped anywhere. The transfer is bound to a variable that should be kept around.

> Is your attempt at bounce buffers working right now?

No, I am currently battling with interrupts. As an experiment, I tried reading the interrupt flags of a DMA channel, and I am able to see the flags change, but if I register an interrupt handler with the low-level API, it never gets called. Not sure what I'm doing wrong on my end.

If you wanna chat about it I'm in the matrix server as @Limeth:matrix.org or @limedyevt on Discord.

EliteTK, 2026-09-06:
@yanshay I've not had a chance to run your code yet, although I still might.

That being said, I've made major progress on my own project in this regard. I have 33 FPS (what you get when you drive this display right in the middle of its frequency range) full depth, full resolution, double-buffered glitch-free output without relying on XiP from PSRAM. But you do need to run the flash at 80MHz which probably won't work reliably without adjusting the timing[^psram-timing].

I allocate two framebuffers in PSRAM, I allocate two 1/16th bounce buffers in RAM. I set up a looping DMA transfer from the two bounce buffers to the LCD peripheral. The special trick I do is that I use 8x more descriptors than necessary, the descriptors alternate between the BBs. But by having descriptors which are in-sync with the screen, I can use the `DMA_OUT_CHx` interrupt to read ` OUT_EOF_DES_ADDR` which is enough information to figure out which exact screen slice just got emitted. This tells me exactly which portion of the current front buffer in PSRAM must be copied into the respective BB. I use Mem2Mem DMA to copy from PSRAM since doing it with the CPU is glacial (reads via cache, and then writes out via cache again) and ends up taking a huge chunk of the frame budget. If the Mem2Mem transfer is still running by the time the next BB must be refilled, I cancel it. Waiting at this point would likely just lead to cascading glitches. This happens very rarely.

Most importantly, even if PSRAM stalls, or you enter a long critical section (`esp-println`'s default features can cause this) then you won't get _persistent_ misalignment/glitches unless somehow the RAM -> LCD transfer stalls which I've not seen happen yet regardless of what has been going on.

The remaining obstacles are just how slow it is to write to PSRAM. A clear of a framebuffer takes milliseconds even with 64 byte cache lines, I am thinking I might use another M2M to do that, in order to avoid the cache and so I can do it eagerly after each frame. Double buffering eliminates screen tearing but complicates damage handling. Although there are plenty of optimisations left.

Either way, very promising.

[^psram-timing]: I'm working on a patch to implement auto-tuning for the 80MHz path, but you might have luck configuring the fallback timings that `esp-idf` uses in case it fails to tune, as I've found these to lie within my chips good range:
    ```
    let spi = esp_hal::peripherals::SPI0::regs();
    let (din_mode, din_num, extra_dummy) = (4, 0, 2);
    
    spi.spi_smem_timing_cali()
        .modify(|_, w| w.spi_smem_timing_clk_ena().set_bit());
    
    unsafe {
        spi.spi_smem_din_mode().modify(|_, w| {
            w.spi_smem_din0_mode().bits(din_mode);
            // ... din1 through din6 ...
            w.spi_smem_din7_mode().bits(din_mode);
            w.spi_smem_dins_mode().bits(din_mode)
        });
    
        spi.spi_smem_din_num().modify(|_, w| {
            w.spi_smem_din0_num().bits(din_num);
            // ... din1 through din6 ...
            w.spi_smem_din7_num().bits(din_num);
            w.spi_smem_dins_num().bits(din_num)
        });
    
        spi.spi_smem_timing_cali().modify(|_, w| {
            w.spi_smem_timing_cali().bit(extra_dummy != 0);
            w.spi_smem_extra_dummy_cyclelen().bits(extra_dummy)
        });
    }
    ```

yanshay, 2026-09-06:
I also implemented rendering slint using a similar method quite some time ago so I don't remember all details by now and it was working quite well (perfect fron UI pov), however due to the flash contention and high PSRAM bandwidth it slowed the application down (I am using PSRAM heavily) and it felt in responsiveness enough that I didn't like it for that application so didn't switch device and still use an I8080 display at lower size and resolution.

MabezDev, 2026-10-27:
The hardware cannot signal transfer completion for DPI, and #6391 adds the alternative `DpiTransfer::wait_for_vsync` in blocking and async mode: https://github.com/esp-rs/esp-hal/commit/2836d02d90fc05a8911e4e6679d6b526a2b7234f

closed 2026-10-27 by MabezDev