esp-rs/esp-hal #2885: Trait bounds were not satisfied when using multiple, configurable, SpiDevices

adoble, 2025-01-04:
## Bug Description

I’m trying to build a ESP32-C3 system with one shared SPI bus connected to many devices. Each of the devices uses a different speed for the data transfer and has its own CS pin. I’m also using the async versions as I want the data transfer to eventually work in embassy tasks.   A condensed version of the code (based on [this](https://docs.embassy.dev/embassy-embedded-hal/git/default/shared_bus/asynch/spi/index.html)) is (`xcs` and `xdcs` are the CS pins):

```rust
#![no_main]
#![no_std]

/// A minimal example to test out how to configure different SpiDevices on a
/// shared spi bus within an embassy context. Using an ESP-C3 as platform
use embassy_embedded_hal::shared_bus::asynch::spi::SpiDeviceWithConfig;
use embassy_executor::Spawner;
use embassy_sync::blocking_mutex::raw::NoopRawMutex;
use embassy_sync::mutex::Mutex;
use embedded_hal_async::spi::Operation;
use embedded_hal_async::spi::SpiDevice;

use esp_backtrace as _;
use esp_hal::gpio::{Level, Output};
use esp_hal::prelude::*;
use esp_hal::spi::master::{Config, Spi};

use esp_hal::timer::timg::TimerGroup;
use static_cell::StaticCell;

type SharedSpiBus = Mutex<NoopRawMutex, Spi<'static, esp_hal::Async>>;

#[esp_hal_embassy::main]
async fn main(spawner: Spawner) {
    let mut peripherals = esp_hal::init({
        let mut config = esp_hal::Config::default();
        config.cpu_clock = CpuClock::max();
        config
    });

    let timg0 = TimerGroup::new(peripherals.TIMG0);

    esp_hal_embassy::init(timg0.timer0);

    let sclk = Output::new(peripherals.GPIO5, Level::Low);
    let mosi = Output::new(peripherals.GPIO6, Level::Low);
    let miso = Output::new(peripherals.GPIO7, Level::Low);
    let xcs = Output::new(peripherals.GPIO9, Level::Low);
    let xdcs = Output::new(peripherals.GPIO10, Level::Low);

    let spi_bus: Spi<'_, esp_hal::Async> = Spi::new(peripherals.SPI2)
        .with_sck(sclk)
        .with_mosi(mosi)
        .with_miso(miso)
        .into_async();

    static SPI_BUS: StaticCell<SharedSpiBus> = StaticCell::new();
    let spi_bus = SPI_BUS.init(Mutex::new(spi_bus));

    let spi_sci_config = Config {
        frequency: 250.kHz(),
        ..Default::default()
    };

    let spi_sdi_config = Config {
        frequency: 8000.kHz(),
        ..Default::default()
    };

    let spi_sci_device = SpiDeviceWithConfig::new(spi_bus, xcs, spi_sci_config);
    let spi_sdi_device = SpiDeviceWithConfig::new(spi_bus, xcs, spi_sdi_config);

    spi_sci_device
        .transaction(&mut [Operation::Write(&[0x01, 0x10])]) // -> Does not compile
        .await
        .unwrap();

    spi_sdi_device
        .transaction(&mut [Operation::Write(&[0x02, 0x20])]) // -> Does not compile
        .await
        .unwrap();
}

```

I get the following compilation error: 

```
error[E0599]: the method `transaction` exists for struct `SpiDeviceWithConfig<'_, NoopRawMutex, Spi<'static, Async>, Output<'_>>`, but its trait bounds were not satisfied
   --> src/main.rs:64:10
    |
63  | /     spi_sci_device
64  | |         .transaction(&mut [Operation::Write(&[0x01, 0x10])]) // -> Does not compile
    | |         -^^^^^^^^^^^ method cannot be called due to unsatisfied trait bounds
    | |_________|
    |
    |
   ::: C:\Users\T440s\.cargo\registry\src\index.crates.io-6f17d22bba15001f\embassy-embedded-hal-0.2.0\src\shared_bus\asynch\spi.rs:115:1
    |
115 |   pub struct SpiDeviceWithConfig<'a, M: RawMutex, BUS: SetConfig, CS> {
    |   ------------------------------------------------------------------- doesn't satisfy `_: SpiDevice<_>`
    |
   ::: C:\Users\T440s\.cargo\registry\src\index.crates.io-6f17d22bba15001f\esp-hal-0.22.0\src\spi\master.rs:462:1
    |
462 |   pub struct Spi<'d, M, T = AnySpi> {
    |   --------------------------------- doesn't satisfy `_: SpiBus<_>`
    |
    = note: the full type name has been written to 'C:\Users\T440s\Documents\rust-projects\spi-device-with-config-esp32c3\target\riscv32imc-unknown-none-elf\release\deps\spi_device_with_config_esp32c3-aa942ddfe493ca7b.long-type-15104742997144927313.txt'
    = note: consider using `--verbose` to print the full type name to the console
    = note: the following trait bounds were not satisfied:
            `esp_hal::spi::master::Spi<'static, Async>: SpiBus<_>`
            which is required by `embassy_embedded_hal::shared_bus::asynch::spi::SpiDeviceWithConfig<'_, NoopRawMutex, esp_hal::spi::master::Spi<'static, Async>, Output<'_>>: embedded_hal_async::spi::SpiDevice<_>`
``` 

After a conversation on the embassy matrix chat group I’m raising this as an issue (although I’m still not ruling out a problem with my implementation). 

## To Reproduce

Build the code using the environment shown at the end.  

## Expected behavior

I would expect it to compile.

## Environment 

Target device is a ESP32-C3 (in my case a DEV-KIT-M1). I’m using rust version 1.83.0 and as the embassy team is in the middle of a new release, I’ve attached the `Cargo.toml` file:

```toml
[package]
name    = "spi-device-with-config-esp32c3"
version = "0.1.0"
edition = "2021"

[dependencies]
embassy-embedded-hal = "0.2.0"
embassy-executor = "0.6.3"
embassy-sync = "0.6.1"
esp-backtrace = { version = "0.14.2", features = [
    "esp32c3",
    "exception-handler",
    "panic-handler",
    "println",
] }
esp-println = { version = "0.12.0", features = ["esp32c3", "log"] }
esp-hal = { version = "0.22.0", features = ["esp32c3"] }
esp-hal-embassy = { version = "0.5.0", features = [
    "esp32c3",
    "log",
    "integrated-timers",
] }

embedded-hal-async = "1.0.0"

static_cell = "2.1.0"
```

Dominaezzz, 2025-01-04:
You'll have to wait for https://github.com/esp-rs/esp-hal/issues/2497 to be released

bugadani, 2025-01-04:
Oops, I asked for this issue because I forgot that I added the implementation 🙈 sorry, will be part of next release.

bjoernQ, 2025-01-06:
closed by #2497