esp-rs/esp-hal #2908: hil-tests might stall at `assert_eq!` etc.

bjoernQ, 2025-01-08:
Repro:

Force a test to fail on `assert_eq!` - e.g.
```diff
@@ -46,6 +46,7 @@ mod tests {
         block_buf[..plaintext.len()].copy_from_slice(plaintext);

         let mut block = block_buf.clone();
+        block[0] = 0;
         ctx.aes.process(&mut block, Mode::Encryption128, keybuf);
         assert_eq!(block, encrypted_message);
     }
```

And run it - e.g. `cargo xtask run-tests esp32c6 -t aes` - execution will stall after printing the error

using `defmt::assert_eq!` either directly or by adding a `use defmt::*;` before `use super::*;`  makes it work as expected

bugadani, 2025-01-08:
Yeah for some reason assert_eq doesn't work with big(ish?) arrays very well. I'm not sure if this is a probe-rs issue (i.e. the test ends up timing out, or something like that), we already have issues with semihosting-printing out messages getting cut off.

MabezDev, 2025-05-02:
> Yeah for some reason assert_eq doesn't work with big(ish?) arrays very well

I believe this is due to the defmt buffer size, I think it hangs if a single item ends up being larger than the buffer. Maybe for HIL tests we can increase the size here?

bugadani, 2025-05-02:
> I believe this is due to the defmt buffer size, I think it hangs if a single item ends up being larger than the buffer.

This is fixed in probe-rs master, might be worth retesting - but I think I still saw the issue recently so maybe this isn't the root cause