esp-rs/esp-hal #2928: Reconnecting an output signal to a different pin does not clear previous connections

bugadani, 2025-01-10:
An example where this matters is SPI hardware chip select. Due to how the IO_MUX works, connecting a new output to the peripheral's (e.g.) CS0 output doesn't clear the old selection, e.g. trying to connect a single SPI chip select signal to multiple pins one by one, then asserting that chip select ends up selecting multiple SPI devices.

bugadani, 2025-01-10:
We don't do this, because we'd have to loop through all GPIOs to find which ones use the signal we want to connect. We could store the last assigned pin in memory, but what if the user might want to connect a signal to multiple pins at once?

Dominaezzz, 2025-01-10:
The GPIO signal semantics haven't been well defined and should probably be sorted out before stabilizing the any drivers that use the GPIO matrix or io mux.

I think the optimal behaviour would be for the driver to store whatever peripheral input/output the user provides, and deconfigure them on drop. In practice this might be annoying to implement.

Looping through all the GPIO won't achieve this because it will also undo any connections setup outside the driver, which is undesirable.

MabezDev, 2025-01-10:
We can logically OR all the pin masks that a driver uses and track the pins that way, but that would mean wasting at least a `u32` but more than likely a `u64` on most chips for each driver. Maybe that's not so bad?

bugadani, 2025-01-10:
> We can logically OR all the pin masks that a driver uses and track the pins that way, but that would mean wasting at least a u32 but more than likely a u64 on most chips for each driver. Maybe that's not so bad?

Heh, good idea, better than to waste a u64 for each signal that I had in mind. Can be part of the driver State.

bugadani, 2025-01-10:
Wait nevermind we still have to know _which_ pin is which signal, we don't want to accidentally reassign MISO as CS0

Dominaezzz, 2025-01-10:
> Wait nevermind we still have to know _which_ pin is which signal, we don't want to accidentally reassign MISO as CS0

That, we can figure out from the GPIO registers 😄

So we store a bitmap of all the output pins the driver owns then use the registers to figure which signal they're mapped to. 

Though I can't tell why we need this information to deconfigure them

PR #3012 (2025-01-23): Introduce PinGuard and reconnecting an output signal to a different pin now clears previous connections
closes #2928
closed 2025-01-23 by MabezDev