esp-rs/esp-hal #2932: Program reading PWM signals with GPIO interrupts weird behavior

tommasoclini, 2025-01-10:
## Bug description

This is the [project](https://github.com/tommasoclini/esp_pwm_cap_rust.git).
The goal is to read the TON of two (possibly more than two) PWM signals coming from an RC receiver (so in us, TON: [1000; 2000], TOFF: [18000; 19000]).
As of now, the program works fine for a while, but then the first channel stops working, in the interrupt handler the call to `input_mut.interrupt_set()` is always false in the case of the first channel (when it stops working).

## To Reproduce

1. Flash an esp32s3 with the provided program
2. Attach two PWM signals of the mentioned kind to GPIOs 14 and 13.

## Expected behavior

Both channels should work.

## Environment

- Target device: ESP32-S3 
- Crate name and version: esp-hal 0.22.0

tommasoclini, 2025-01-10:
What can I do to try to see what's going on? How do I catch what apparently is unlistening the interrupt for the first channel?

tommasoclini, 2025-01-10:
If I call listen in the interrupt handler whenever the interrupt flag is not set (it shouldn't make sense to do this), the problem disappears, what could the problem be?
I think it's just the fact that I'm calling listen fast enough(because of the other channel that stays alive) to make moments where it gets disabled very short.

tommasoclini, 2025-01-12:
I've written a new [version](https://github.com/tommasoclini/esp_pwm_cap_rust/tree/f0ecfec3749d739ceb6e3975556940a62d0c78e5) of the program, a button press calls listen on every channel.
I've noticed that most of the time, if not always, one of the two channels stops working whenever the TON value is very similar, I've got a video for demonstration.
[Screencast from 2025-01-12 16-50-01.webm](https://github.com/user-attachments/assets/d4016445-4ccb-4018-95e5-c85a6007db78)
Whenever a channel stop, if I press the button, it starts working again, where(or when?) could be the issue here?

tommasoclini, 2025-01-12:
@MabezDev, any idea?

tommasoclini, 2025-01-12:
I've written a similar [program](https://github.com/tommasoclini/esp_pwm_cap_cpp.git) in c++, I'm not facing the problem in that case, the logic is also a bit different, should I try using a single handler like I do in the rust program?

tommasoclini, 2025-01-13:
Thank you, I'll think about it. Also, what could be causing the rust program to stop working correctly? The c++ version has no problems.
I've insisted with using GPIOs only because in the past I had made a similar program in c++ and it was reading 4 pwm channels no problem.

tommasoclini, 2025-01-13:
I tried setting ESP_HAL_PLACE_ANON_IN_RAM to true, it didn't change anything, unfortunately.

tommasoclini, 2025-01-13:
Also, I'd like to get minimum latency for each period, so I kind of have to be able to receive a TON as soon as it takes place, is the RMT peripheral really helping me in the case where I want to capture single events?

tommasoclini, 2025-01-13:
Interesting.
[image]
0 corresponds to the press of the button that enables the interrupt again.
14 corresponds to the channel being taken down.

Where could the issue be? It seems like my handler sometimes doesn't clear the interrupt in time, and gets detected by the async driver, and rightfully shut down.

liebman, 2025-01-13:
Another option would be to use the capture module of the `MCPWM` peripheral.  There are two `MCPWM` units on the `esp32s3` and each has 3 capture channels.  This will give you timestamped signal edges.

tommasoclini, 2025-01-13:
Is that supported in this hal?

tommasoclini, 2025-01-13:
I see that my issue has probably been fixed in the main branch, should I just switch to that instead of using the crates.io version?

liebman, 2025-01-13:
Depending on what features you need you may need to write the registers directly (my last use was in my pre-rust days)

tommasoclini, 2025-01-13:
I don't really have time for that, I'm a solo dev working on the firmware of a 1:10 racing car (we have a competition in May, the team is @PFI-E-Racing) this would mean a big chunk of time just to read the channels of a PWM receiver, if I can get it done with the fix in the main branch, I'll definitely do that.

bjoernQ, 2025-01-14:
> I see that my issue has probably been fixed in the main branch, should I just switch to that instead of using the crates.io version?

Worth a try. There will be also a 0.23.0 release, soon.

In general, using a dedicated peripheral would be good since you won't need the CPU involved but if GPIO + interrupts work good enough for you (assuming it works with main/0.23.0) it's probably fine

MabezDev, 2025-01-14:
@tommasoclini As stated multiple times, you _will_ be better off using dedicated hardware for this. With that said, you have a final option to improve latency by binding directly to CPU interrupts instead of going through the HAL dispatcher. See https://docs.esp-rs.org/esp-hal/esp-hal/0.22.0/esp32c6/esp_hal/interrupt/fn.enable_direct.html. This does come with added complexity, and will likely break async GPIO unless you call the handler yourself, but it's there as an option.

You now have two or three ways to solve this, I don't consider this a HAL bug anymore. Closing.

closed 2025-01-14 by MabezDev

bjoernQ, 2025-01-14:
> @bjoernQ when is 0.23 expected to be released?

It's planned for end of this week (if nothing unforeseen happens)