esp-rs/esp-hal #2939: Full-duplex mode of I2S is not working/implemented yet.

foldingbeauty, 2025-01-13:
## Bug description

I'm writing a driver for the SGTL5000 audio codec chip and during testing I found out I can't read and write from one I2S peripheral. I looked at the code and I can see that the code  reading and writing the right channels of de I2S is implemented the right way and I can read or write but not both.  I looked at the code and I can see @bjoernQ (great work!) wrote most of it, if this is not a bug but a feature, I'm happy to contribute to the feature, but I need some pointers for the contribution to be effective. 

## To Reproduce

1. Use the following code:
``` rust
// SGTL5000 need a MCLK
i2s = i2s.with_mclk(peripherals.GPIO16);

// Setup the rx channel on the I2S peripheral.
let mut i2s_rx = i2s.i2s_rx
        .with_bclk(peripherals.GPIO6)
        .with_ws(peripherals.GPIO4)
        .with_din(peripherals.GPIO15)
         build();

// Setup the tx channel on the  I2S peripheral, I can't specify the bclk/ws because the value is moved.   
let mut i2s_tx = i2s.i2s_tx
        .with_dout(peripherals.GPIO5)
        .build();
```
3. With this code the rx part is working.
4. When I change the code into:
``` rust
// SGTL5000 need a MCLK
i2s = i2s.with_mclk(peripherals.GPIO16);

// Setup the rx channel on the I2S peripheral
let mut i2s_rx = i2s.i2s_rx
         .with_din(peripherals.GPIO15)
         .build();

// Setup the tx channel on the I2S peripheral    
       let mut i2s_tx = i2s.i2s_tx
            .with_bclk(peripherals.GPIO6)
            .with_ws(peripherals.GPIO4)
            .with_dout(peripherals.GPIO5)
```
5. The transmit channel is working but the receiving side of things doesn't work. 

## Expected behavior

The expected behaviour would be that both transmitting and receiving works or/but  maybe the code should be like this (which is not implemented this way):

```
i2s = i2s.with_mclk(peripherals.GPIO16).with_bclk(...).with_ws(...);
// Setup the receive channel on the I2S peripheral
let mut i2s_rx = i2s.i2s_rx
         .with_din(peripherals.GPIO15)
         .build();
// Setup the transmit channel on the I2S peripheral
let mut i2s_tx = i2s.i2s_tx
         .with_dout(peripherals.GPIO15)
         .build();

```

## Environment

- Target device: ESP32-S3
- Crate name and version: esp-hal 0.22.0

bjoernQ, 2025-01-13:
Thanks!

We currently use full-duplex for loopback testing in our HIL tests - maybe that helps: https://github.com/esp-rs/esp-hal/blob/7d7ce279d139bfd306f2dee586c198f1821bbf15/hil-test/tests/i2s.rs#L81-L97

The driver doesn't support an API for that, yet.

foldingbeauty, 2025-01-13:
@bjoernQ thanks for the quick reply.  Will take a look soon. I got all the information I need.

foldingbeauty, 2025-01-17:
@bjoernQ I did a a lot of research into the code, ESP32S3 programming guides en looked at all the registers being used and how the pins are assigned. I can conclude that Full-duplex is out-of-the-box enabled by your code. After tweaking the codec I got it working without changing the code. The programming guides said if you set the I2S to master mode and the RX and TX config channels (not the DMA but the I2S channels) are assigned to the same pins you got full-duplex. And you already implemented this way.

I think you can close this issue as keep it as a reference. 

With the configuration I created a  test sample looper application and mix a couple of recording and send it back to the codec:

```rust

    i2s = i2s.with_mclk(peripherals.GPIO4);
    let mut i2s_tx = i2s
        .i2s_tx
        .with_bclk(peripherals.GPIO5)
        .with_ws(peripherals.GPIO6)
        .with_dout(peripherals.GPIO7)
        .build();

    let mut i2s_rx = i2s.i2s_rx
        .with_bclk(NoPin)
        .with_ws(NoPin)
        .with_din(peripherals.GPIO9)
        .build();

```

bjoernQ, 2025-01-17:
Thanks for the update!

main--, 2026-04-18:
Up until #5292 was filed, this issue was (as far as I could find) the only place describing how to run duplex I2S with esp-hal. I don't know if the S3 is an exception here, but I want to clarify to anyone finding this in the future that the code in https://github.com/esp-rs/esp-hal/issues/2939#issuecomment-2597931261 is **not** a working general solution. The TX side works fine, but as reported in #5292 you end up with randomly shifted data on the RX side, as the RX clock you are actually sampling DIN on is just getting sent into the void.

QuackHack-McBlindy, 2026-04-24:
I want to thank you Björn for the help and support - I finally got it to work.
The funny thing is that I actually had it right all along - it was the speaker task that was not providing clocks when audio was not being played, had to write zero when audio is not played.

> I double checked and can't find them on `crates.io` - but I'd appreciate seeing them released.

Oh, I must have messed something up, I will try to get them released this weekend!
Thanks again!

bjoernQ, 2026-04-26:
Thanks for  the update! Awesome to hear it's working fine now!

QuackHack-McBlindy, 2026-05-25:
Has anyone had time to do through testing of the `signal_loopback` yet?

I have been getting  some critical interference and have soon ruled out everything but the loopback. Running half-duplex without the loopback give crystal clear sound.
At first I thought the interference was caused by a sample‑rate mismatch between the configured sample rate and the actual hardware clock that the ESP32 can generate. (by counting samples length over time). But after more testing I noticed that was also the cause when running half-duplex without issues.

https://github.com/QuackHack-McBlindy/share/blob/main/noise.wav

The noise is not an hardware issue as I have been getting it from all my devices. 
Might come from my codebase but I have been unable to confirm that either.
The interference might sound minimal, but it's rendering speech-to-text barely usable and seems non trivial to remove using hacky-code solutions.