esp-rs/esp-hal #2949: Consider making the MSRV of each package consistent

jessebraham, 2025-01-14:
Currently each package has a unique MSRV, based purely on the requirements of the given package. This is usually determined via tooling, e.g. [cargo-msrv](https://github.com/foresterre/cargo-msrv).

With the MSRV of `esp-hal` (and all packages which depend on it) now set to 1.83, I'm proposing that we increase the MSRV of the remaining packages to match.

1. Our MSRV checks in CI are currently not even checking most of the packages in this repository, as we would need to specify a large number of Rust versions and it would be a headache. Having a consistent MSRV would allow us to actually check this. Hard to say we're providing any guarantees when we are not checking the supposed MSRV.
2. None of these packages are intended for use with any project outside of the `esp-hal` ecosystem to my knowledge, so I'm not sure what advantages there are to having an MSRV lower than that of `esp-hal`.

Thoughts?

bjoernQ, 2025-01-14:
> None of these packages are intended for use with any project outside of the esp-hal ecosystem to my knowledge, so I'm not sure what advantages there are to having an MSRV lower than that of esp-hal.

+1 ... I think technically some packages could be used w/o the HAL but I doubt anyone is doing that

bugadani, 2025-01-14:
I'd propose moving straight to 1.84 and enabling the new cargo resolver that came with it immediately. This is just mildly relevant, but if we touch the MSRV once, we might as well go to the latest immediately.

closed 2025-01-14 by jessebraham