esp-rs/esp-hal #2970: esp-hal-embassy low_power_wait  cannot be enabled

relufi, 2025-01-15:
## Bug description
Failed to compile after low power wait is enabled

## To Reproduce
```toml
[env]
ESP_HAL_EMBASSY_CONFIG_LOW_POWER_WAIT = "false"
```
```
error[E0412]: cannot find type `AtomicBool` in this scope                                          
  --> C:\Users\relufi\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\esp-hal-embassy-0.6.0\src\executor\thread.rs:18:34
   |
18 | static SIGNAL_WORK_THREAD_MODE: [AtomicBool; Cpu::COUNT] =
   |                                  ^^^^^^^^^^ not found in this scope
   |
help: consider importing one of these structs
   |
3  + use core::sync::atomic::AtomicBool;
   |
3  + use portable_atomic::AtomicBool;
   |

error[E0433]: failed to resolve: use of undeclared type `AtomicBool`
  --> C:\Users\relufi\.cargo\registry\src\rsproxy.cn-0dccff568467c15b\esp-hal-embassy-0.6.0\src\executor\thread.rs:19:14
   |
19 |     [const { AtomicBool::new(false) }; Cpu::COUNT];
   |              ^^^^^^^^^^ use of undeclared type `AtomicBool`
   |
help: consider importing one of these structs
   |
3  + use core::sync::atomic::AtomicBool;
   |
3  + use portable_atomic::AtomicBool;
   |

```

## Expected behavior

## Environment

- Target device:  ESP32-S3
- Crate name and version: esp-hal-embassy 0.6.0

relufi, 2025-01-15:
https://github.com/esp-rs/esp-hal/blob/4b66e3dba736f30ca370f670a8d3e1012fde50a3/esp-hal-embassy/src/executor/thread.rs#L9-L20

bugadani, 2025-01-15:
Thanks! For now you can spawn a task that just yields in a loop, as a workaround. Fixing this properly will need #2971 to be resolved so we can actually test the fix.

MabezDev, 2025-01-20:
Closed via https://github.com/esp-rs/esp-hal/pull/2975, we can track adding the test(s) in #2971