esp-rs/esp-hal #2984: ESP32-C6: Using an L2CAP MTU < 255 with BLE causes a runtime error

lure23, 2025-01-16:
## Bug description

In the [trouble](https://github.com/embassy-rs/trouble) examples, MTU values of 128 and 251 are being used. When such examples are being run on ESP32-C6 hardware, the following error occurs:

```
INFO - [host] setting txq to 24, fragmenting at 255
INFO - [host] configuring host buffers (8 packets of size 251)

====================== PANIC ======================
panicked at /home/ubuntu/trouble.fork/examples/apps/src/ble_bas_peripheral.rs:104:13:
[ble_task] error: BleHost(Hci(Invalid HCI Command Parameters))
```

@bjoernQ [asked me to report](https://github.com/embassy-rs/trouble/pull/236#issuecomment-2586457641) this here.

According to him, also ESP32-C2 and ESP32-H2 might be affected (I have only ESP32-C6 to test with).

## To Reproduce

1. clone `trouble`
2. edit `examples/esp32/Cargo.toml`, disabling the workaround feature:

```
[features]
...
esp32c2 = ["esp-hal/esp32c2", "esp-backtrace/esp32c2", "esp-hal-embassy/esp32c2", "esp-println/esp32c2", "esp-wifi/esp32c2"]
esp32c3 = ["esp-hal/esp32c3", "esp-backtrace/esp32c3", "esp-hal-embassy/esp32c3", "esp-println/esp32c3", "esp-wifi/esp32c3"]
esp32c6 = ["esp-hal/esp32c6", "esp-backtrace/esp32c6", "esp-hal-embassy/esp32c6", "esp-println/esp32c6", "esp-wifi/esp32c6"]
esp32h2 = ["esp-hal/esp32h2", "esp-backtrace/esp32h2", "esp-hal-embassy/esp32h2", "esp-println/esp32h2", "esp-wifi/esp32h2"]
```

Run with:

```
$ cargo esp32c6 --bin ble_bas_peripheral
```

## Expected behavior

Output (starts advertising):

```
...
INFO - [adv] advertising
```

## Actual behaviour

There's a panic at initialization of the BLE stack.

## Environment

- Target device: ESP32-C6, -C2, -H2

   ```
   Chip type:         esp32c6 (revision v0.0)
   Crystal frequency: 40 MHz
   Flash size:        4MB
   Features:          WiFi 6, BT 5
   ````

- Crate name and version: `esp-hal` rev. `7288d0367e0a820db4ca25a22496a35beff1bb61`

bjoernQ, 2025-01-16:
Thanks!

We need to make parts of `BLE_CONFIG`  (e.g. https://github.com/esp-rs/esp-hal/blob/660e43effaa1ef6a0419db7c77a5034485ce9c41/esp-wifi/src/ble/os_adapter_esp32c3.rs#L249 and https://github.com/esp-rs/esp-hal/blob/660e43effaa1ef6a0419db7c77a5034485ce9c41/esp-wifi/src/ble/os_adapter_esp32c6.rs#L16) user configurable

Unfortunately, it's different for different chips

lure23, 2025-02-09:
According to [this](https://matrix.to/#/#esp-rs:matrix.org/$ZzC-QWHocidnEXtn5vAxcsGnUDLTnk4NGf9Cr7kVrjo) on Matrix, also ESP32-S2 [is affected](https://matrix.to/#/#esp-rs:matrix.org/$KY-upuyXCab_St5lPxMoAqB7z5tOtT_21KAESlzCkHw).