esp-rs/esp-hal #3008: Potential deadlock on esp32-c3 when using listening sockets on embassy-net

TethysSvensson, 2025-01-21:
## Bug description

When running an echo-server using 4 sockets over wifi, sometimes my program will freeze up under high load in what appears to be a deadlock.

The last lines of logging appears to come from the function `lock_mutex` inside the file `esp-wifi/src/compat/common.rs`.

## To Reproduce

I have a reproducer here: https://github.com/TethysSvensson/esp32-wifi-deadlock-poc

## Expected behavior

I do not a expect my board to freeze up.

## Environment

```
$ espflash board-info
[2025-01-21T13:42:57Z INFO ] 🚀 A new version of espflash is available: v3.3.0
[2025-01-21T13:42:57Z INFO ] Serial port: '/dev/ttyACM0'
[2025-01-21T13:42:57Z INFO ] Connecting...
[2025-01-21T13:42:57Z INFO ] Using flash stub
Chip type:         esp32c3 (revision v0.4)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BLE
MAC address:       40:4c:ca:8c:f4:0c
```

I am using `esp-hal` version `0.23.1`. For other versions, see my reproducer.

TethysSvensson, 2025-01-21:
Is this code being called from inside a critical section? If so, then that would explain why the `yield_task` doesn't do anything.

bjoernQ, 2025-01-21:
The Rust code is this

https://github.com/esp-rs/esp-hal/blob/537829df3635b701fc8de695245e10411a06ee30/esp-wifi/src/compat/common.rs#L281-L309

So `yield_task` is not called from within the CS

`42010cec:	30047673          	csrrc	a2,mstatus,8` would clear bit 3 (MIE = Global machine mode interrupt enable)

`42010d1e:	30082073          	csrs	mstatus,a6`would enable interrupts (given A6 is contains 8)

This should raise the interrupt to yield
```asm
42010d26:	03432583          	lw	a1,52(t1) # 600c0034 <_rtc_fast_bss_end+0x100c0034>
42010d2a:	0015e593          	or	a1,a1,1
42010d2e:	02b32a23          	sw	a1,52(t1)
```

So, the code here looks correct - maybe that's a red herring and something else is happening (and the debugger lied to me). Luckily, we have the reproducer

TethysSvensson, 2025-01-21:
I meant that perhaps the `lock_mutex` function is called from a critical section (despite clearly not being intended to do so).

bjoernQ, 2025-01-22:
> I meant that perhaps the `lock_mutex` function is called from a critical section (despite clearly not being intended to do so).

Sorry, my bad. Yes, seems to be the case since MIE is 0 in that situation.

bjoernQ, 2025-01-22:
Interesting observaton.

Replacing `esp_hal::sync::Locked` with this
```rust
pub struct Locked<T> {
    data: critical_section::Mutex<RefCell<T>>,
}

impl<T> Locked<T> {
    /// Create a new instance
    pub const fn new(data: T) -> Self {
        Self {
            data: critical_section::Mutex::new(RefCell::new(data)),
        }
    }

    /// Provide exclusive access to the protected data to the given closure.
    ///
    /// Calling this reentrantly will panic.
    pub fn with<R>(&self, f: impl FnOnce(&mut T) -> R) -> R {
        critical_section::with(|cs|{
            let mut x = self.data.borrow_ref_mut(cs);
            f(&mut x)
        })
    }
}
```

in esp-wifi makes things work again. (At least it's running now for approx. one hour)

<details>

<summary>DIFF</summary>

```diff
diff --git a/esp-wifi/src/compat/common.rs b/esp-wifi/src/compat/common.rs
index 219ac01e..01c3931d 100644
--- a/esp-wifi/src/compat/common.rs
+++ b/esp-wifi/src/compat/common.rs
@@ -12,11 +12,11 @@ use esp_wifi_sys::include::malloc;
 use super::malloc::free;
 use crate::{
     binary::c_types::{c_int, c_void},
-    hal::sync::Locked,
     memory_fence::memory_fence,
     preempt::current_task,
     timer::yield_task,
 };
+use crate::Locked;
 
 pub(crate) const OSI_FUNCS_TIME_BLOCKING: u32 = u32::MAX;
 
diff --git a/esp-wifi/src/compat/timer_compat.rs b/esp-wifi/src/compat/timer_compat.rs
index 2c98529a..7b8cf59b 100644
--- a/esp-wifi/src/compat/timer_compat.rs
+++ b/esp-wifi/src/compat/timer_compat.rs
@@ -1,7 +1,7 @@
 use alloc::boxed::Box;
 
-use esp_hal::sync::Locked;
 
+use crate::Locked;
 use crate::binary::{
     c_types,
     include::{esp_timer_create_args_t, ets_timer},
diff --git a/esp-wifi/src/lib.rs b/esp-wifi/src/lib.rs
index 697176b3..857bc327 100644
--- a/esp-wifi/src/lib.rs
+++ b/esp-wifi/src/lib.rs
@@ -100,7 +100,7 @@ extern crate alloc;
 // MUST be the first module
 mod fmt;
 
-use core::marker::PhantomData;
+use core::{cell::RefCell, marker::PhantomData};
 
 use common_adapter::chip_specific::phy_mem_init;
 use esp_config::*;
@@ -481,3 +481,32 @@ fn init_clocks() {
     let mut radio_clocks = unsafe { esp_hal::peripherals::RADIO_CLK::steal() };
     radio_clocks.init_clocks();
 }
+
+
+
+
+
+
+
+pub struct Locked<T> {
+    data: critical_section::Mutex<RefCell<T>>,
+}
+
+impl<T> Locked<T> {
+    /// Create a new instance
+    pub const fn new(data: T) -> Self {
+        Self {
+            data: critical_section::Mutex::new(RefCell::new(data)),
+        }
+    }
+
+    /// Provide exclusive access to the protected data to the given closure.
+    ///
+    /// Calling this reentrantly will panic.
+    pub fn with<R>(&self, f: impl FnOnce(&mut T) -> R) -> R {
+        critical_section::with(|cs|{
+            let mut x = self.data.borrow_ref_mut(cs);
+            f(&mut x)
+        })
+    }
+}
\ No newline at end of file
diff --git a/esp-wifi/src/preempt/mod.rs b/esp-wifi/src/preempt/mod.rs
index 1589fb99..746d8fea 100644
--- a/esp-wifi/src/preempt/mod.rs
+++ b/esp-wifi/src/preempt/mod.rs
@@ -5,7 +5,7 @@ pub mod arch_specific;
 use core::mem::size_of;
 
 use arch_specific::*;
-use esp_hal::sync::Locked;
+use crate::Locked;
 use esp_wifi_sys::include::malloc;
 
 use crate::{compat::malloc::free, hal::trapframe::TrapFrame, memory_fence::memory_fence};
diff --git a/esp-wifi/src/timer/mod.rs b/esp-wifi/src/timer/mod.rs
index 56fd1dd9..c2721ceb 100644
--- a/esp-wifi/src/timer/mod.rs
+++ b/esp-wifi/src/timer/mod.rs
@@ -1,4 +1,4 @@
-use esp_hal::sync::Locked;
+use crate::Locked;
 
 #[cfg_attr(esp32, path = "timer_esp32.rs")]
 #[cfg_attr(esp32c2, path = "timer_esp32c2.rs")]
diff --git a/esp-wifi/src/wifi/event.rs b/esp-wifi/src/wifi/event.rs
index ac997674..7dd11526 100644
--- a/esp-wifi/src/wifi/event.rs
+++ b/esp-wifi/src/wifi/event.rs
@@ -1,6 +1,6 @@
 use alloc::boxed::Box;
 
-use esp_hal::sync::Locked;
+use crate::Locked;
 
 use super::WifiEvent;
 
diff --git a/esp-wifi/src/wifi/mod.rs b/esp-wifi/src/wifi/mod.rs
index aede5895..88106d3a 100644
--- a/esp-wifi/src/wifi/mod.rs
+++ b/esp-wifi/src/wifi/mod.rs
@@ -13,7 +13,7 @@ use core::{
 };
 
 use enumset::{EnumSet, EnumSetType};
-use esp_hal::sync::Locked;
+use crate::Locked;
 use esp_wifi_sys::include::{
     esp_eap_client_clear_ca_cert,
     esp_eap_client_clear_certificate_and_key,
```

</details>

TethysSvensson, 2025-01-22:
Here is my current understanding of the problem:

* We are seeing a hang in the function [`fn lock_mutex(mutex: *mut c_void) -> i32`](https://github.com/esp-rs/esp-hal/blob/ae35b465bf4ea7dee4330df8c9a3d60e98f3aa96/esp-wifi/src/compat/common.rs#L282).
* This function is only called through the wrapper [`unsafe extern "C" fn mutex_lock(mutex: *mut crate::binary::c_types::c_void) -> i32`](https://github.com/esp-rs/esp-hal/blob/ae35b465bf4ea7dee4330df8c9a3d60e98f3aa96/esp-wifi/src/wifi/os_adapter.rs#L350)
* This wrapper function is stored in the [`g_wifi_osi_funcs` table](https://github.com/esp-rs/esp-hal/blob/ae35b465bf4ea7dee4330df8c9a3d60e98f3aa96/esp-wifi/src/wifi/mod.rs#L1271) and in the [`G_OSI_FUNCS` table](https://github.com/esp-rs/esp-hal/blob/ae35b465bf4ea7dee4330df8c9a3d60e98f3aa96/esp-wifi/src/ble/os_adapter_esp32c3.rs#L108).
* The `G_OSI_FUNCS` table is passed as an argument to the `esp_register_ext_funcs`/`btdm_osi_funcs_register` functions in `esp-wifi/src/ble/npl.rs`/`esp-wifi/src/ble/btdm.rs`. These functions are native functions defined in `libble_app.a`/`libbtdm_app.a`. Since they are bluetooth specific, this is unlikely to be directly relevant to the current issue, but might have a similar problem.
* The `g_wifi_osi_funcs` is a global extern referenced by `libnet80211.a` and `libwpa_supplicant.a`. It is also stored in the [`G_CONFIG` table](https://github.com/esp-rs/esp-hal/blob/ae35b465bf4ea7dee4330df8c9a3d60e98f3aa96/esp-wifi/src/wifi/mod.rs#L1406), which is passed to `esp_wifi_init_internal`. This function is defined in `libnet80211.a`.
* The surface-level reason why we're seeing this hang seems to be because `MIE`, so sometimes (perhaps always?) this function gets called with `MIE` set to 0?
* One avenue of investigation is to see how this is handled in the idf.
* Here we can also find [the `g_wifi_osi_funcs` table](https://github.com/espressif/esp-idf/blob/0f0068fff3ab159f082133aadfa9baf4fc0c7b8d/components/esp_wifi/esp32c3/esp_adapter.c#L594), which also has a [similar wrapper function](https://github.com/espressif/esp-idf/blob/0f0068fff3ab159f082133aadfa9baf4fc0c7b8d/components/esp_wifi/esp32c3/esp_adapter.c#L196).
* Two differences spring to mind: the wrapper function is stored in the `.iram1` section and inner function has an argument used as a timeout.

I'm not sure if any of this is helpful, but I might as well share the research I've done.

bugadani, 2025-01-22:
> Replacing esp_hal::sync::Locked with this

Locked should be virtually equivalent with `Mutex<RefCell<T>>` on single core. It is implemented using critical section and a boolean flag, it just panicks if you take it reentrantly - but so does the RefCell if you access it mutably twice.

A bug in Locked is somewhat worrying :/

bjoernQ, 2025-01-22:
While I was writing above comment it stalled for me (but as said after approx. one hour)

> The surface-level reason why we're seeing this hang seems to be because MIE, so sometimes (perhaps always?) this function gets called with MIE set to 0?

Usually, MIE is 1 here. The implementation of critical-section is modifying it. I start to see MIE=0 not before the second client connects to the reproducer. When doing the above mentioned change I don't see MIE=0 in that function anymore.

(BTW it seems that just using critical-section instead of Lock for `DATA_QUEUE_RX_*` is enough to get over the initial lock-up - not too surprising since that is what is used with a critical-section by the network stack)

bjoernQ, 2025-01-22:
> Locked should be virtually equivalent with `Mutex<RefCell<T>>` on single core. It is implemented using critical section and a boolean flag, it just panicks if you take it reentrantly - but so does [...]

Indeed - this observation if worrying me a lot! (I wish I haven't seen this happening 🙈 )

bjoernQ, 2025-01-22:
Seems I found (and fixed the issue). Good news is it's nothing mystical happened - just a plain old bug (hidden / masked for too long)

PR #3015 (2025-01-22): esp-wifi: fix possible deadlock
## Thank you for your contribution!

We appreciate the time and effort you've put into this pull request.
To help us review it efficiently, please ensure you've gone through the following checklist:

### Submission Checklist 📝
- [ ] I have updated existing examples or added new ones (if applicable).
- [x] I have used `cargo xtask fmt-packages` command to ensure that all changed code is formatted correctly.
- [x] My changes were added to the [`CHANGELOG.md`](https://github.com/esp-rs/esp-hal/blob/main/esp-hal/CHANGELOG.md) in the **_proper_** section.
- [ ] I hav [... cut]