esp-rs/esp-hal #3013: [RFC] Should the `next` field in the `DMADescriptor` be an `Option<NonNull<DMADescriptor>>`?

Frostie314159, 2025-01-21:
Hi, while looking through the docs for the `DMADescriptor` I found that the `next` field is a `*mut DMADescriptor`. This is valid, since there are no guarantees, that the pointer is not null, which it will be if there are no further elements. However, it might be more idiomatic and easier to use, if this would instead be a `Option<NonNull<DMADescriptor>>`. This should be sound as well, since according to the docs for [`std::option`](https://doc.rust-lang.org/std/option/index.html#representation) NPO is guaranteed for this case, and a `transmute` is sound. Doing this would probably help prevent misuse.
If you want to, I can make a PR for this, in which I'd also add a few unsafe convenience methods to deal with the associated buffer.

bugadani, 2025-01-22:
We don't really have a strong preference here, but in my experience NonNull tends to be somewhat awkward. If you think it would be cleaner than using the raw pointes, I don't think we'll stop you from sending in a PR :)

Dominaezzz, 2025-01-22:
My only concern is this for loop (and others like it) will look slightly more complex https://github.com/esp-rs/esp-hal/blob/67c915917648245d3a3dc23d57f478cf908dd331/esp-hal/src/dma/buffers.rs#L1174

Frostie314159, 2025-01-22:
Ok, sounds good. I think I can find a way to make those loops not too complex.

closed 2025-01-22 by Frostie314159